gyroflow/gyroflow #287: Unusable audio export

BabaTom, 2022-03-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

Rc 1.05

### What operating system are you using?

Mac OS 11.6.5 Big Sur

### What GPU are you using?

2.3 GHz 8-Core Intel Core i9

### What happened?

This latest build appears to allow one to export audio, yet the audio is a garbled screechy mess, not the nice ambiance and dialogue that I filmed. Original files I am working with are .mxf files from a Sony FX6. Audio camera records is 48khz linear PCM. I understand that, as currently designed, Gyroflow should pass thru channel 1 when exporting. What I saw is that my stabilized files had actually passed thru all 8 audio channels that FX6's write. What would be smart and ideal would be if Gyroflow would simply pass whatever audio channels are on an existing file thru and export them - unmolested (!). As it is now, its a bit of a hassle having to re-attach audio from my unstabilzed files to the gyroflow stabilized ones. Great program that serves a great purpose overall, thank you! Tom
PS Would attach a relevant, outputted from Gyroflow, file but it's 2gb

BabaTom, 2022-03-28:
here you go, sending to you via WeTransfer, it is an h264 file but was same result in ProRes HQ

BabaTom, 2022-03-28:
Download link https://we.tl/t-mT4Zjsg6iz

1 item
581_0350_stabilized.mp424.6 MB

mycosd, 2022-03-29:
Thanks for the file, but I actually meant an unstabilized version that we can run through Gyroflow for testing.

BabaTom, 2022-03-30:
Dear Maik:

Sorry slow reply; under gun here today, alas.

Untouched-from-camera file is at this link (25fps 4K Sony FX6 S-log 3 cine), shot on Zeiss Milvus 18mm):

Download link 
https://we.tl/t-zNNzrY63qR <https://we.tl/t-zNNzrY63qR>

Shot this little clip as a test, to see if I can get Gyroflow to eliminate this nasty panning jitter, but discovered the audio problem in the process. I was unable to fully eliminate this nasty jitter using Gyroflow, but maybe that’s my incompetence (!). Attaching the profile I made for this lens in case anyone there may also want to test stabilization with it.

Best, Tom

_______________________
Thomas Dallal
www.thomasdallal.com
***@***.***
+972 54 981 1854
+972 59 764 3931
+1 646 801 9640

mycosd, 2022-03-30:
Thanks for the video. As you mentioned the jitter, I just opened the file and noticed the gyro/auto sync information actually looking fine. So there is no reason it couldn't be stabilized. However auto-sync doesn't work on that file (this might be another issue to look into), but you can place a manual sync point and for me the result looks quite smooth with a sync offset of -44ms. You can use the optical flow lines (auto-sync generates them) and align them with the gyro graphs.

[image]

mycosd, 2022-04-04:
Audio issue should be fixed in the latest dev-build: https://gyroflow.xyz/devbuild/?autodownload

closed 2022-04-04 by mycosd

BabaTom, 2022-04-05:
Wow. Thank you for flying me in, Maik. As I am a software neophyte, to be kind, maybe I should wait for rc6 to test I there? If I download the dev build, will it replace the RC5 I have installed? And when rc6 arrives, will that then replace all else? Or, pardon my ignorance, should I be uninstalling each as the builds progress? If so, how is best on a Mac?

Thanks. Tom

Ps. How is best to contribute to this excellent work you all are doing?

_______________________
Thomas Dallal
www.thomasdallal.com
***@***.***
+972 54 981 1854
+972 59 764 3931
+1 646 801 9640

mycosd, 2022-04-05:
> As I am a software neophyte, to be kind, maybe I should wait for rc6 to test I there? If I download the dev build, will it replace the RC5 I have installed? And when rc6 arrives, will that then [...]

I'm not very familiar with Mac OS, you should be able to have multiple versions on your system simultanously. @AdrianEddy correct me if I'm wrong.

BTW: Elvin & Adrian also fixed the auto-sync issue mentioned above. So this might be another reason to use the dev-build version.

> Ps. How is best to contribute to this excellent work you all are doing?

You can help by spreading the word about Gyroflow, or donate here: https://gyroflow.xyz/donate/

AdrianEddy, 2022-04-05:
> If I download the dev build, will it replace the RC5 I have installed? 
It won't replace anything by itself, you can just extract dev build to a different directory, it doesn't have to be `Applications`, and you can have multiple versions in different locations or just with other names

BabaTom, 2022-04-07:
Thank you, Maik and Adrien.

Just tested out latest Dev Build with five clips I filmed yesterday. Amazing: (1) all 8 of the default FX6 audio tracks now pass thru when I export stabilized footage as ProRes HQ, which is exactly how it should be, and; (2) Auto-sync worked nicely on my S-log 3 cine log files - I had to manually add a sync point to 2 files that were especially flat looking (though they were not plain walls or anything that extreme) but no big deal, definitely much improved, and by now I see a pattern of needing to subtract something in the -41 or so milliseconds at one point for a short clip, which seems to give quite a good result. Maybe I will get more sophisticated with this, I hope. Footage was not very wobbly but I still had a tad of l-r yaw in some of it, will experiment further when I have more time.

I sent you a modest contribution (USD 25) and will share your existence with a fellow FX6 user pal here in the parts who will surely be thrilled.

Thanks again, guys. You’re on to something great with this, in my humble estimation.

Kind regards,

Tom
_______________________
Thomas Dallal
www.thomasdallal.com
***@***.***
+972 54 981 1854
+972 59 764 3931
+1 646 801 9640