gyroflow/gyroflow #288: "

BabaTom, 2022-03-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

Rc 1.05

### What operating system are you using?

Mac OS 11.6.5 Big Sur

### What GPU are you using?

2.3 GHz 8-Core Intel Core i9

### What happened?

When I go to export (as ProRes HQ) a gyroflow synched and stabilized FX6 clip (using a properly done lens profile, btw) from gyroflow, I get a message saying there are no sync points on the clip and asking if program should proceed. Please see attached screenshot. THere does not seem to be any issue with exported file but I do wonder if Gyroflow is doing its best with my files or if mayve there may be a fly in the ointment working with FX6 .mxf files (which are h264, of course). Never had this message with RC 1.04, with which I also used for files with same exact camera and lens. See attached screenshot. Thanks, Tom
[image]

AdrianEddy, 2022-03-28:
Your gyro is not synchronized, stabilization will not be correct.
You need to have at least one sync point (yellow marker on the timeline).

Closing this as it's not an issue but intended behavior and important warning

closed 2022-03-28 by AdrianEddy

AdrianEddy, 2022-10-11:
Autosync doesn't mean that the sync points are actually created. It analyzes the video, but if it can't find corresponding movement in the gyro data, it won't create sync points. There are a couple of possible reasons for that:
- Too small "Sync search size"
- Incorrect IMU orientation
- Incorrect gyro rotation (if the gyro source is not pointing in the same direction as the camera)
- Too much noise in the gyro data
- Invalid rolling shutter value
- Not accurate lens profile

I see you're using FX6 with internal gyro data. In this case, this list reduces to:
- Not accurate lens profile
- Too much noise in the gyro data
- Invalid rolling shutter value