gyroflow/gyroflow #289: fake motion blur on stabilized output

CREASE-gum-EAR, 2022-03-29:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

During stabilization, could motion blur be generated to match? blur comes from the motion and with new motion should be new motion blur to match? If this was an option, I would fly with a high framerate with a fast shutter for crisp video and then add motion blur back into the footage matching the new stabilization. We could then dial in the motion blur after the fact and it would look perfect.

ghuser123321, 2022-04-06:
That's what I do. Fly with high shutter, then use Resolve to add motion blur to taste. There seems to be a way even in the free version: https://beginnersapproach.com/davinci-resolve-add-motion-blur/

CREASE-gum-EAR, 2022-04-06:
Yeah I was aware of motion blur software doing optical flow calcs in order to do motion blur. But with gyro data, we can get "better" results? I wasn't sure how much this would help make it easier to calculate.

AdrianEddy, 2022-04-06:
You still need to do it based on optical flow, motion blur is not simply camera motion, it's really an object motion visually in the frame, so you can't use gyro for that.
For example imagine flying straight very fast and passing a building - the horizon won't have motion blur, the building should have strong motion blur, and gyro data indicates basically no movement.

I doubt this will be implemented in Gyroflow, there are already existing solutions for that, I recommend _RE:Vision Effects ReelSmart Motion Blur Pro_

CREASE-gum-EAR, 2022-04-14:
Ahhh yeah that's right @AdrianEddy so this is really not appropriate for gyroflow. I'll close this comment then. Sorry to be a distraction.

closed 2022-04-14 by CREASE-gum-EAR