gyroflow/gyroflow #294: Add Max Zoom Level

craigy72, 2022-03-30:
When using Gyro flow for FPV plane, during launch there can be a rotation of around 90-180 degrees of the aircraft. The result of this is that gyro flow, before launch zooms in alot (maybe 200 - 300 pc), then launch and it zooms out. This design is unpleasable to the eyes. 

It would be great to have a parameter which is max zoom setting that would be enable across the entire video. Or, the user could apply a max zoom to a specific area of the video.

mycosd, 2022-03-30:
I don't think that we can actually implement something like that in GF at the moment. The dynamic zoom is calculated from the smoothing output, so it just zooms in when there is not enough of the image left to draw the rotation coming from the smoothing.
However I think you can achieve the same result once #37 is implemented, by setting keyframes on the "smoothing amount" so at the beginning there is basically minimal amount of smoothing and after the take-off you can increase it.

AdrianEddy, 2022-03-30:
For this specific use case, you can just set the trim range to exclude the take off and landing and it will not use this portion when calculating dynamic zoom

craigy72, 2022-03-31:
Thanks I will  have a look at using Trim range. There might be other use cases to trim not just the start / end but also in the middle of the video - not sure if tri can do this. i.e. it sounds like its just the start and end?

AdrianEddy, 2022-04-19:
This will be possible once we implement #37 so I'll close this one. You can click "Subscribe" in #37 to get notified once it's implemented