gyroflow/gyroflow #298: Allow custom FFMPEG output or FFMPEG frame serving

ghuser123321, 2022-04-05:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

First of all let me thank you for all your time dedication that made this project possible. It's incredible!

Currently export codecs and configuration are limited to a few choices. Different users have different needs. Please allowing advanced users to set a custom FFMPEG line in a textbox to configure export codecs exactly the way they require.

If that's not possible, please consider frame serving to a custom FFMPEG binary.

Use cases for me:

* Only rate control option currently is bitrate, which is an understandable, but IMHO a sub optimal choice. Constant Quality with custom quantization parameters makes much more sense in this day and age (to me).
* Allow lossless h264 (8bit) and lossless h256 (10) 4:2:0 and 4:4:4 output using NVENC with custom keyframe interval (-g option in FFMPEG). Such files with short GOP sizes are good to work with in Resolve, provide maximum quality and still result in 2-3x space savings over image sequences.

I'm  envisioning a textbox where I could input "-c:a:0 pcm_s24le -c:v hevc_nvenc -preset lossless -pix_fmt yuv444p10" and hit export.

AdrianEddy, 2022-04-05:
The problem is that we don't use ffmpeg CLI so there's no way to just pass these options, everything is called directly using C API of ffmpeg.
`pix_fmt`, `preset` and `gop` distance could be added without too much changes (but will complicate the UI even more), but other than that, I don't think it makes sense to implement custom frame passing to ffmpeg binary - it would require to rewrite half of the rendering pipeline. 

BTW: DNX output will probably be added soon

ghuser123321, 2022-04-06:
Thanks for your suggestion. That would be amazing and help me a lot. Gyroflow is so fast on a 3090 that any sort CPU encoding is the bottleneck and slows it down to a fraction of possible speed.

GPU encoding on the other hand has so little options (bitrate) that I can't use it (to produce easily editable high quality intermediate files for Resolve). I tried hard to use the OFX version, but IMHO it's not ready yet. I'll file separate issues for that.

I think the UI is one of the most straight forward and simple interfaces I've seen in a long time. Hiding those options behind an "advanced" button wouldn't create clutter?

Having
* pix_fmt
* preset
* gop distance
* selection for bitrate or constant quality rate control 

would be a dream and speed up my workflow multiple times.

AdrianEddy, 2022-05-05:
I'll add to that list ability to select x264/x265 tier and level

AdrianEddy, 2022-06-11:
Done in: fa4dbaac512936db039f5390e8d0ab3d3f0858c3
https://gyroflow.xyz/devbuild/?autodownload

closed 2022-06-11 by AdrianEddy

AdrianEddy, 2022-07-22:
Setting keyframe distance to 0s should set it to 1 frames
You should be able to set these parameters almost directly, just change `--rc:v:0 vbr` to `-rc vbr` and remove qmin and qmax
qmin and qmax are separate APIs and not implemented in Gyroflow

User-3090, 2022-07-22:
Thanks. I probably wasn't completely clear above.

`-profile:v rext -pix_fmt yuv444p16le --rc:v:0 vbr -tune hq -cq 0 -qmin 0 -qmax 99`

appears to work just fine in Gyroflow. I'm now confused as you say -qmin and -qmax has no effect as they clearly have an effect when used within the above line. `-qmax 12` produces a file 4 times larger than `-qmax 99` in Gyroflow when used as above.

0s seems to work for all-intra and this what I mostly need.

Still, IMHO setting keyframe distance is preferable in frames rather than seconds for the reason outlined above. What do you think?

AdrianEddy, 2022-07-22:
Which encoder are you using? `hevc_nvenc'? Also drop all :v from the options, it won't work with it, just have `-profile rext`

User-3090, 2022-07-22:
Yes, hevc_nvenc.

Both lines produce an identical file when used inside Gyroflow and in both variants changing -qmax changes file size dramatically.

`-profile:v rext -pix_fmt yuv444p16le --rc:v:0 vbr -tune hq -cq 0 -qmin 0 -qmax 99`
`-profile: rext -pix_fmt yuv444p16le -rc vbr -tune hq -cq 0 -qmin 0 -qmax 99`

AdrianEddy, 2022-07-22:
Interesting, then it must be handled internally by ffmpeg. There are separate fields for qmin and qmax in the API and i don't set or change them

User-3090, 2022-07-22:
Also I find that `-pix_fmt` seems single threaded and CPU bound and bottle-necking render speed to about 1/3 of possible speed. But this probably more of an ffmpeg issue. Do you have control over ffmpeg threading from withing Gyroflow?

[image]

User-3090, 2022-07-22:
> Interesting, then it must be handled internally by ffmpeg. There are separate fields for qmin and qmax in the API and i don't set or change them

it still complains about adding -g1, though :p

AdrianEddy, 2022-07-22:
Keep in mind that this is not a FFmpeg command line, these are only encoder parameters.

Pixel format conversion is done on the Gyroflow side (by calling ffmpeg API) and it's single threaded. The plan is to eventually replace ffmpeg conversion with GPU conversion for common formats

AdrianEddy, 2022-07-22:
Is there a reason to convert to 4:4:4 16-bit? You won't gain any additional quality versus using original pixel format

User-3090, 2022-07-23:
Okay, can you help me understand why that is?

I was opting for 4:4:4 as Gyroflow applies a complex transformation and I want to give it full chroma resolution to do so. Are you saying that if input format is 4:2:0 Gyroflow will internally do the transformation only at 4:2:0 chroma resolution and all that -pix_fmt does is later in the chain let ffmpeg upscale chroma before encoding?

I don't need or want 16 bit HEVC. `-pix_fmt yuv444p16le` is just the only way I found to get 4:4:4 10bit HEVC from nvenc_hevc. `yuv444p16le` is truncated to 10bit.

I'm aware there is `p016le` which is supposedly the correct way to get 4:4:4 10bit HEVC from nvenc_hevc, but it doesn't work for me. For me it produces 4:2:0 files similar to `p010le`. I found this info [here](https://superuser.com/a/1623636).

AdrianEddy, 2022-07-23:
> I was opting for 4:4:4 as Gyroflow applies a complex transformation and I want to give it full chroma resolution to do so. Are you saying that if input format is 4:2:0 Gyroflow will internally do [...]

Yes, Gyroflow works on separate YUV planes, and it's not doing any rescaling of chroma for processing. If the input file is 4k 4:2:0 10-bit, it will process the Y plane at 3840x2160 resolution, and then it will process U and V planes at 1920x1080 resolution (this is how the pixels are stored). And if you set `pix_fmt`, the conversion is done later, after processing by Gyroflow. It is possible to make the processing after the conversion, but it's pointless and adds unnecessary scaling up of U and V planes.

User-3090, 2022-07-23:
Is it actually pointless? If you are doing a fish eye correction you'll have more source pixels crammed into less space in the center. If this processing is done at source resolution you'll lose details in the middle as those pixels need to somehow be averaged to fit into less space. If processing/canvas space would be larger you wouldn't loose information.

What is wrong with my thinking?

AdrianEddy, 2022-07-23:
But these pixels are not added magically, they are interpolated from source pixels. When you do the upscaling from 4:2:0 to 4:4:4 you still need to use an interpolation of the source pixels. Then you do the second interpolation in stabilization processing to correct fisheye and move pixels around. Both are currently Lanczos4 interpolation. In your case, the interpolation is done twice. Without pixel format conversion it's done only once.

User-3090, 2022-07-24:
Okay, I think I understood. Also with that knowledge it's obviously non-sense what I'm currently doing.

But: As long as whatever Gyroflow produces is not the final delivery and goes into an NLE, most of them (Resolve and Vegas, not Premiere) will anyway do another chroma interpolation. And they do so using crappy bilinear (Vegas) or variant thereof (Resolve) math.

So if Gyroflow could use Lanczos to produce 4:4:4 in a single interpolation step that would be a big improvement. And sort of what I was hoping would happen when setting pix_fmt to 4:4:4.

AdrianEddy, 2022-07-24:
This is possible to implement in theory, but in reality it's a real pain in the ass and horrible to maintain. Sure we can do single  YUV 10-bit 4:2:0 to 4:4:4, but do you know how many pixel formats are out there? For the most basic ones, we would have to support at least 15-20, and all of them will have different math to process that. Implementing that without repeating all code for all pixel formats is really challenging and not something I would like to spend my time on. In the end, the quality gain from something like this will be minimal, if at all noticeable.