gyroflow/gyroflow #299: Edges come into frame with some lens profiles when setting lens correction strength to zero

avsaase, 2022-04-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

gh416

### What operating system are you using?

Windows 7

### What GPU are you using?

AMD Radean RX 570

### What happened?

With some lens profiles, black areas come into the frame when using dynamic zoom and setting the lens correction strength to 0:

[image]

This example is from the Runcam 5 Orange with the 4k@30fps(XV) setting and a lens profile I made with the new input stretch setting. The problem also occurs with the Runcam Thumb and the lens profile shipped with Gyroflow so I don't think it's related to input stretch. 

Sample video, gyro logs and lens profile are available [here](https://drive.google.com/drive/folders/1ogzt9rqXUBWdFq3mRLb_j8BVPbMKhAo2?usp=sharing).

avsaase, 2022-04-05:
Hmm, the same video with lens correction strength at 100% also gets some edges in view
[image]

AdrianEddy, 2022-04-05:
It should be fixed in latest dev build (gh418), please confirm and close

avsaase, 2022-04-05:
With the latest dev build the problem is indeed fixed. However, with lens correction strength set to 0%, it zooms in way more than it needs to. For example, with smoothness set to the minimum, this is how far it zooms in:

[image]

AdrianEddy, 2022-04-05:
and if you set zooming speed also to minimum?

avsaase, 2022-04-05:
That only has a marginal effect on the zoom level.

avsaase, 2022-04-09:
Hmm, with the `vertical_input_stretch` lens profile you posted on Discord there is no unnecessary zoom. It took me a while to notice because when switching lens profiles you also need to drag the lens correction strength slider a bit to get it to take full effect.

Pavelo77, 2022-04-16:
This happens also with Linear Profile Gopro videos, and is especially strong on the second chapter part when two files are joined with Reelsteady Joiner: the black frame comes in strong way before the turn, like the processing is out of sync. I also eperienced out of sync gyro data in the second chapter with RC5, when joined or not (not fixable with autosync) but that's fixed in the last dev build.

AdrianEddy, 2022-04-16:
@Pavelo77 please share the video file

Pavelo77, 2022-04-16:
Here are the videos, shot in Linear Mode 4k 60p on a GoPro 8 black, stabilized with latest dev buil 13042022.
Second chapter part by itself: https://youtu.be/VemmKVo-PPE
Same second chapter part in a joined video: https://youtu.be/LtKfdL-ZmFg
Black borders come in both, way stronger in the second one.

AdrianEddy, 2022-04-16:
@Pavelo77 please share the original video file via google drive/dropbox/wetransfer or similar, I need to load it in gyroflow to reproduce the issue

Pavelo77, 2022-04-16:
Sure, here you go:
https://drive.google.com/file/d/1fwFfFm9Of818a3qiMoEPCutbfSv_iGS9/view?usp=sharing
https://drive.google.com/file/d/1vBh9--P04AYEUjRepr4r4NAUg7Xj_fP4/view?usp=sharing
I joined them with the latest Reelsteady Joiner 1.2.4

AdrianEddy, 2022-04-30:
> Hmm, with the `vertical_input_stretch` lens profile you posted on Discord there is no unnecessary zoom. It took me a while to notice because when switching lens profiles you also need to drag the lens correction strength slider a bit to get it to take full effect.
Dynamic zoom for XV profile is now fixed in 717e3447f92d168405c586fecce0e8a179cf57c9

AdrianEddy, 2022-04-30:
@Pavelo77 your issue is fixed in d6c6dcde4387b958c77d3f4752fe604ac61e6a66
Download latest dev build here: https://gyroflow.xyz/devbuild/?autodownload (gh464)

closed 2022-04-30 by AdrianEddy

Pavelo77, 2022-05-01:
it is, thank you very much!!