gyroflow/gyroflow #300: Allow render from command line

ghuser123321, 2022-04-06:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

I'm aware Gyroflow started as an FPV tool to process single long files. However it's so amazing that many adopt it (and I'm advocating for it in different forums) for other use cases. One of these use cases is to basically use it on anything shot handheld or to smooth gimbal work further.

That means many short per-scene files to open and configure settings, then wait for render before working on the next file. It would really help if we could get a command line switch to trigger a render for a gyroflow configuration file.

Something like

`Gyroflow.exe -render somefile.gyroflow`

That way we could batch renders overnight with a shell loop. I'm aware there is an issue to improve the .gyroflow file to include render settings, so this is probably a dependency.

AdrianEddy, 2022-04-13:
This will be done in #27

closed 2022-04-13 by AdrianEddy

AdrianEddy, 2022-04-21:
You can now drag&drop `.gyroflow` files to the render queue, this was added in cec97ec1f750b5e85f4246034827fd0e854d4c7c

ghuser123321, 2022-04-21:
This is so cool. Thank you so much!

AdrianEddy, 2022-09-01:
Done in 3c4219d2eb4b4f8c3ee02b72ba9b3abc2dc08dc2 and 1ff6506219683b35e90f1a076b80895eccbe5289
Available in latest dev build: https://gyroflow.xyz/devbuild/?autodownload
```
Usage: gyroflow.exe [<input...>] [-f] [-j <parallel-renders>] [-d <when-done>] [-p <out-params>]

Positional Arguments:
  input             input files: videos, project files, lens profiles, presets

Options:
  -f, --overwrite   overwrite if output file exists, default: false
  -j, --parallel-renders
                    number of parallel renders, default: 1
  -d, --when-done   when done: 1 - shut down; 2 - reboot; 3 - sleep; 4 -
                    hibernate; 5 - logout
  -p, --out-params  output parameters, eg. "{ 'codec': 'H.265/HEVC', 'bitrate':
                    150, 'use_gpu': true, 'audio': true }"
  --help            display usage information
```

Basically provide video or project file, and optionally you can specify lens profile file and/or preset file. It supports rendering multiple files in a single call.

If any functionality is missing, let me know your use case.