gyroflow/gyroflow #301: GoPro Hero 10: Horizon lock flips 180 degree on acceleration / decceleration or direction changes with

ghuser123321, 2022-04-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

gh418

### What operating system are you using?

Win 10

### What GPU are you using?

RTX 3090

### What happened?

GoPro 10 placed on gimbal with roll and pitch axis fixed / stabilized. On quick acceleration or direction changes (e.g. running, chasing something) it repeatedly flips 180 degree over back and forth.

ghuser123321, 2022-04-12:
Sorry for the delay, if more sample footage is required please let me know and I'll upload. Forget to mention that the GoPro is mounted 180 degree flipped on my gimbal at all times.

ghuser123321, 2022-04-12:
Yes, I have exactly the same issues as [Marcus-Zhu](https://github.com/Marcus-Zhu). It also starts 90 degree rotated and then flips back and forth.

AdrianEddy, 2022-05-02:
@User-3090 and @Marcus-Zhu - this should be fixed in latest dev build for Hero 9 and 10, please check. https://gyroflow.xyz/devbuild/?autodownload

Hero 8 doesn't have gravity vector in metadata so it probably won't work, but you can also try

User-3090, 2022-05-02:
It's fixed and works great!

But one thing: Could you pre-populate the "Roll angle correction" with the "Rotation" angle from the video information panel, please? Else the video flips for everyone with non-zero rotation angle as soon as Lock horizon is enabled.

Thanks a lot!

closed 2022-05-02 by AdrianEddy