gyroflow/gyroflow #302: Make loop behaviour consistent (always?)

ghuser123321, 2022-04-07:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Currently it only seems to loop when a trim section is selected and the section end is not the clip end. Can we have it loop always?

AdrianEddy, 2022-04-08:
@wang-bin How to loop at end when `player->setProperty("continue_at_end", "1");` is set?

wang-bin, 2022-04-09:
try the latest build

ghuser123321, 2022-04-12:
I tried gh425 and behavior has not changed as far as I can tell:
* No loop when no trim range exists
* No loop when trim range exists but ends with clip end
* Loops when trim range exists and its end is before clip end

AdrianEddy, 2022-04-12:
@ghuser123321 wang-bin meant the mdk-sdk build, not gyroflow build, it's not updated yet in gyroflow. You can replace the mdk by yourself by downloading from https://sourceforge.net/projects/mdk-sdk/files/nightly/ and replacing in gyroflow directory. 

I didn't test it myself yet

ghuser123321, 2022-04-12:
Thanks. It's fixed.

There is still some sort of play head jump or slowdown when it reaches the last frames of the clip, but I'd consider that a cosmetic issue :)

ghuser123321, 2022-04-13:
Fixed.

closed 2022-04-13 by ghuser123321