gyroflow/gyroflow #306: Experimental zero-copy GPU preview often fails when opening new file

ghuser123321, 2022-04-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

gh425

### What operating system are you using?

Win 10

### What GPU are you using?

Strix 3090

### What happened?

Experimental zero-copy GPU preview fails often opening new files or dragging new file inside the GUI. It then shows a single random color. Almost if it would zoom in so that it shows only one pixel of the video. The state then sticks until restart of Gyroflow.

Also the setting is not saved/persisted making it harder to test.

AdrianEddy, 2022-04-12:
I know, that's why it's still experimental :) And it's not saved in settings on purpose, so people don't accidentally enable it and face issues without realizing it was this setting

AdrianEddy, 2022-04-28:
Let's keep track of it in #40

closed 2022-04-28 by AdrianEddy

AdrianEddy, 2022-09-29:
It should finally be stable in latest dev build https://gyroflow.xyz/devbuild/?autodownload