gyroflow/gyroflow #307: reset defaut value

Francesco-Roma, 2022-04-12:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Add a new feature for reset all the value of the application as the initial value of the first installation

AdrianEddy, 2022-04-12:
Windows:
1. Windows+R and type `regedit` or look for `Registry Editor` in Start menu
2. Go to `Computer\HKEY_CURRENT_USER\SOFTWARE\Gyroflow` (enter this in the top bar and hit Enter)
3. Delete the entire key

MacOS:
Open Terminal and enter: `rm -f $HOME/Library/Preferences/com.gyroflow-xyz.Gyroflow.plist ; killall cfprefsd`

Linux:
Open Terminal and enter: `rm -f $HOME/.config/Gyroflow/Gyroflow.conf`

closed 2022-04-12 by AdrianEddy

AdrianEddy, 2022-08-24:
In latest Gyroflow version, there's a button to reset all settings in the Advanced section

Francesco-Roma, 2022-08-31:
Big Adrian.

Thank you.

lukegluke, 2022-10-11:
@AdrianEddy is there a possibility to replace default settings with custom ones? For now each time I have to drop setting file just to set usual output resolution and bitrate for my single camera.

AdrianEddy, 2022-10-11:
@lukegluke  in v1.3.0-rc you can put your preset as `default.gyroflow` in `camera_presets` directory and it will be applied to every video you load