gyroflow/gyroflow #310: Stabilization & dynamic zoom not recalculated when changing trim range while playing

ghuser123321, 2022-04-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

gh425

### What operating system are you using?

Win10

### What GPU are you using?

Strix 3090

### What happened?

Trim range can be adjusted while playing, but stabilization and dynamic zoom is not recalculated after the change leading to bogus preview.

Expected behavior: Recalculate stabilization when trim range changed

AdrianEddy, 2022-04-14:
It's not recalculated while you drag the trim range because it's too much calculations and calculating stabilization and dynamic zoom is costly. It's updated when you stop dragging. This is by design.

Preview while dragging is useful to see what exactly you're trimming - lack of dynamic zoom for this brief moment is not a problem

closed 2022-04-14 by AdrianEddy

ghuser123321, 2022-04-14:
Your right, it normally behaves like that. I somehow managed to break that behavior once, but I fail now to reproduce. Will comment if I can come up with a determistic repo.