gyroflow/gyroflow #311: Playhead hard to drag near clip end

ghuser123321, 2022-04-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

gh425

### What operating system are you using?

Win10

### What GPU are you using?

3090

### What happened?

For some reason I find the play head quirky to drag -- in general, but particular if it's close to the clip end. It tends to wildly jump somewhere else when dragged near clip end.

AdrianEddy, 2022-04-14:
Can you show a screen recording of this behavior?

ghuser123321, 2022-04-15:
Sure:
https://user-images.githubusercontent.com/71881005/163563523-3645a872-e476-41f8-a186-8da30bc29604.mp4

ghuser123321, 2022-04-15:
https://user-images.githubusercontent.com/71881005/163563673-09a2b093-5b85-4665-a7da-9da667ae0a45.mp4

AdrianEddy, 2022-04-15:
Hmm, it doesn't behave this way for me, but it will depend on your input file. If you try with a file in intermediate codec like ProRes or DNxHD, you'll see no jumping. The problem is in the way the frames are stored in x264/x265 files and the keyframe distance, some files may have this behavior better and some worse. It's nothing in Gyroflow though, it'll be in ffmpeg or MFT or any underlying decoder which handles the file

AdrianEddy, 2022-04-15:
can you send me that file?

mycosd, 2022-04-17:
I've looked into that one some time ago (happens with my Hero8 HEVC clips too). It somhow comes from the MDK-Player returning weird frame positions while seeking. I don't see this issue related to GOP. In my clips I have a keyframes every second, but the jumps are way bigger and always to the beginning of the trim range.
It also only appear to happen if you have a trim range set.

ghuser123321, 2022-04-18:
You're right. It's within a trim range and not particular to the clip end. It was just coincidence that I often trim at the end of the clip. Anyway, I'll upload sample footage for this and other issues.

AdrianEddy, 2022-07-14:
Is this still an issue in latest dev build? https://gyroflow.xyz/devbuild/?autodownload

mycosd, 2022-07-14:
I can't reproduce it anymore, not even with the version I've noticed the issue in. Not sure what's going on, maybe GPU driver update?

closed 2022-07-18 by AdrianEddy