gyroflow/gyroflow #312: Support for OpenCamera Sensors Fork

Nobodi99, 2022-04-18:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Hello,

I recently saw a video about gyroflow and was instantly intrigued.
After searching arounf and trying a few things I was wondering if it would be possible to support the Sensors Fork of OpenCamera.

It's an Android app that logs the data from the internal sensors while taking a video on the phone which negates the need for another camera. The only downside is that it doesn't record sound but that's not a big problem in my opinion.

I tried to import the generated CSV file into gyroflow but it can't parse the information contained.
The logged information is not that much different from Sensor Logger. it is just ordered differently and missing the "elapsed seconds" part.
Maybe it would be possible to convert the csv outside of gyroflow but I can't seem to find a working solution.

OpenCamera Sensors saves the data like this: X-data, Y-data, Z-data, timestamp (ns)
And here is an example from the start of the csv file. (No headers in file too)

> -0.295013,-0.12901473,0.21542598,8925125055592799
-0.29298842,-0.13505358,0.21694443,8925125057574049
-0.2902657,-0.14224434,0.21788691,8925125059574049
-0.2875779,-0.15053464,0.21921335,8925125061574049
-0.28585002,-0.15959291,0.22101104,8925125063577174
-0.28419197,-0.16926204,0.22270401,8925125065574049
-0.28290042,-0.18109536,0.22560126,8925125067591236
-0.28173104,-0.19507545,0.22992967,8925125069574049
-0.28000316,-0.2086541,0.23453735,8925125071574049

If anyone got an idea to convert this or could implement this I would be really grateful. Buying a camera for the few things I currently film would be too much at the moment :D

AdrianEddy, 2022-04-18:
Please attach full csv file and corresponding video

Nobodi99, 2022-04-18:
[2022_04_18.zip](https://github.com/gyroflow/gyroflow/files/8507636/2022_04_18.zip)

I quickly took a short test clip. I hope that helps?
The zip contains the clip and all sensor data that is logged.
I believe the xxxgyro.csv should contain the necessary data.

Slint7, 2022-06-05:
I also want this feature.

nm17, 2022-09-28:
@AdrianEddy any news on this? I can provide some more demo files

AdrianEddy, 2022-09-28:
The problem is that their files don't have any headers so there's no way to reliably detect them. If they add headers then it will be added

AdrianEddy, 2022-11-14:
Done in https://github.com/AdrianEddy/telemetry-parser/commit/ef1737b8256bd7870ebb0a04d0c2ae1901ca67ad

Stabilized sample by @Nobodi99:

https://user-images.githubusercontent.com/3849067/201714647-b71428cc-0854-4530-98dc-2e626eab8e7e.mp4

closed 2022-11-14 by AdrianEddy

blogdemoi, 2023-03-03:
> Done in [AdrianEddy/telemetry-parser@ef1737b](https://github.com/AdrianEddy/telemetry-parser/commit/ef1737b8256bd7870ebb0a04d0c2ae1901ca67ad)
> 
> Stabilized sample by @Nobodi99:

Do you mean I can use videos recorded with "OpenCamera Sensors" in Gyroflow?

AdrianEddy, 2023-03-03:
yes