gyroflow/gyroflow #313: Gf keeps crashing on launch, no errors

Fede163x, 2022-04-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

rc5, rc4

### What operating system are you using?

windows 10

### What GPU are you using?

nvidia geforce gtx 1050

### What happened?

Gyroflow keeps crashing, without reporting errors, every time i try to open it.
It show only a blank page for about two seconds.
i've already install all updates for graphics, wind and vc.redistr.x64
I've tried rc5 and rc4, same things happened

AdrianEddy, 2022-04-18:
what happens when you open `Gyroflow_with_console.bat`?

Fede163x, 2022-04-18:
it says:
12:14:38 [DEBUG] (2) gyroflow::util: Minidump uploaded: OK
12:14:41 [INFO] Lens profiles directory: "\\\\?\\C:\\Users\\skiop\\Downloads\\Gyroflow-windows64\\camera_presets"
12:14:41 [INFO] Latest version: v1.0.0-rc5, current version: 1.0.0-rc5
12:14:41 [INFO] Loaded 1293 lens profiles in 38.202ms
Crashdump written to C:\Users\skiop\Downloads\Gyroflow-windows64/a2a56ee2-ef78-49e0-824d-2d5c76352e71.dmp
Premere un tasto per continuare . . .

Fede163x, 2022-04-18:
how can i attach the dmp file?

AdrianEddy, 2022-04-18:
Can you try in the dev build? https://gyroflow.xyz/devbuild/?autodownload

Fede163x, 2022-04-18:
same thing happened

12:48:23 [DEBUG] (4) gyroflow::util: Minidump uploaded: OK
12:48:26 [INFO] Latest version: v1.0.0-rc5, current version: 1.0.0-rc5 (gh425)
12:48:27 [INFO] Lens profiles directory: "\\\\?\\C:\\Users\\skiop\\Downloads\\gyroflow\\Gyroflow-windows64\\camera_presets"
12:48:27 [INFO] Loaded 1293 lens profiles in 36.733ms
Crashdump written to C:\Users\skiop\Downloads\gyroflow\Gyroflow-windows64/fd6ef335-3341-4494-be70-d19c4a9c85e3.dmp
Premere un tasto per continuare . . .

AdrianEddy, 2022-04-18:
What's your device? Is that a laptop or PC? do you have two graphic cards? (Intel HD and NVIDIA for example)?

Fede163x, 2022-04-18:
it's a laptop, yes i have both intel hd and nvidia

Fede163x, 2022-04-18:
[image]

AdrianEddy, 2022-04-18:
looks like there's some issue with the Intel HD graphics driver, did you update it?

Fede163x, 2022-04-18:
unistall and reinstall all with the intel assistant, now it works! thank u a lot :)))

closed 2022-04-18 by Fede163x