gyroflow/gyroflow #315: i want to run the project in visual studio code

kxifenger, 2022-04-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.0.0-rc5

### What operating system are you using?

mac os monterey 12.1

### What happened?

how to run this project in visual studio code
i download the project and run in the visual studio code but it do not work.
[image]
i need some help. thank you

AdrianEddy, 2022-04-25:
Follow `Building on MacOS` from the README: https://github.com/gyroflow/gyroflow#building-on-macos
Don't use built-in "Run" and "Debug" options in VS Code, it's hard to setup this

kxifenger, 2022-04-25:
> Follow `Building on MacOS` from the README: https://github.com/gyroflow/gyroflow#building-on-macos Don't use built-in "Run" and "Debug" options in VS Code, it's hard to setup this

thank you for you reply，i want to know how to debug this project，and what IDE use to develop

closed 2022-04-28 by AdrianEddy