gyroflow/gyroflow #317: Building on MacOS

kxifenger, 2022-04-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

master

### What operating system are you using?

mac os monterey 12.1

### What happened?

i got this error when Building on MacOS in step 8 Compile and run: cargo run --release
   Compiling opencv v0.63.1
   Compiling qml-video-rs v0.1.0 (https://github.com/AdrianEddy/qml-video-rs.git#a3f0d56e)
error: failed to run custom build command for `opencv v0.63.1`

Caused by:
  process didn't exit successfully: `/Users/ifeng/gyroflow/target/release/build/opencv-d9c26589bbf64d98/build-script-build` (exit status: 1)
  --- stdout
  running: "/Users/ifeng/.rustup/toolchains/stable-x86_64-apple-darwin/bin/cargo" "build" "--release" "--package" "opencv-binding-generator" "--bin" "binding-generator"
  cargo:rerun-if-env-changed=OPENCV4_NO_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG_x86_64-apple-darwin
  cargo:rerun-if-env-changed=PKG_CONFIG_x86_64_apple_darwin
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG
  cargo:rerun-if-env-changed=OPENCV4_STATIC
  cargo:rerun-if-env-changed=OPENCV4_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_STATIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64-apple-darwin
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64_apple_darwin
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64-apple-darwin
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64_apple_darwin
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64-apple-darwin
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64_apple_darwin
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_SYSROOT_DIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR
  cargo:rerun-if-env-changed=OPENCV_NO_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG_x86_64-apple-darwin
  cargo:rerun-if-env-changed=PKG_CONFIG_x86_64_apple_darwin
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG
  cargo:rerun-if-env-changed=OPENCV_STATIC
  cargo:rerun-if-env-changed=OPENCV_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_STATIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64-apple-darwin
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64_apple_darwin
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64-apple-darwin
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64_apple_darwin
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64-apple-darwin
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64_apple_darwin
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_SYSROOT_DIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR

[... 725 lines cut ...]
  to the PKG_CONFIG_PATH environment variable
  No package 'opencv4' found
  , `"pkg-config" "--libs" "--cflags" "opencv"` did not exit successfully: exit status: 1
  error: could not find system library 'opencv' required by the 'opencv' crate

  --- stderr
  Package opencv was not found in the pkg-config search path.
  Perhaps you should add the directory containing `opencv.pc'
  to the PKG_CONFIG_PATH environment variable
  No package 'opencv' found

  === Probing OpenCV library using cmake
  === cmake ninja probe command: "cmake" "-S" "/Users/ifeng/.cargo/registry/src/github.com-1ecc6299db9ec823/opencv-0.63.1/cmake" "-DOCVRS_PACKAGE_NAME=OpenCV" "-DCMAKE_BUILD_TYPE=Release" "-G" "Ninja"
  === Probing with cmake ninja generator failed, will try Makefile generator, error: No such file or directory (os error 2)
  === cmake makefiles probe command: "cmake" "-S" "/Users/ifeng/.cargo/registry/src/github.com-1ecc6299db9ec823/opencv-0.63.1/cmake" "-DOCVRS_PACKAGE_NAME=OpenCV" "-DCMAKE_BUILD_TYPE=Release" "-G" "Unix Makefiles"
  === Probing with cmake Makefile generator failed, will try deprecated find_package, error: No such file or directory (os error 2)
  === cmake find-package compile probe command: "cmake" "-S" "/Users/ifeng/.cargo/registry/src/github.com-1ecc6299db9ec823/opencv-0.63.1/cmake" "-DOCVRS_PACKAGE_NAME=OpenCV" "-DCMAKE_BUILD_TYPE=Release" "--find-package" "-DCOMPILER_ID=GNU" "-DLANGUAGE=CXX" "-DMODE=COMPILE" "-DNAME=OpenCV"
  === Can't probe using: cmake, continuing with other methods because: No such file or directory (os error 2)
  === Probing OpenCV library using vcpkg_cmake
  === Can't probe using: vcpkg_cmake, continuing with other methods because: Could not find Vcpkg tree: No vcpkg installation found. Set the VCPKG_ROOT environment variable or run 'vcpkg integrate install'
  === Probing OpenCV library using vcpkg
  === Can't probe using: vcpkg, continuing with other methods because: Could not find Vcpkg tree: No vcpkg installation found. Set the VCPKG_ROOT environment variable or run 'vcpkg integrate install', Could not find Vcpkg tree: No vcpkg installation found. Set the VCPKG_ROOT environment variable or run 'vcpkg integrate install'
  Error: "Failed to find OpenCV package using probes: environment, pkg_config, cmake, vcpkg_cmake, vcpkg"
      Finished release [optimized] target(s) in 0.02s
how to solve it

kxifenger, 2022-04-25:
i run  this command   cargo run --release
sometimes i got this error: failed to run custom build command for opencv v0.63.1
sometimes i got this error: failed to run custom build command for ffmepg
i am so confused，how can i build successful

kxifenger, 2022-04-25:
i run this command ./deploy-macos.sh
i got this info
   Compiling png v0.16.8
   Compiling crossbeam-channel v0.5.4
   Compiling core-foundation v0.9.3
   Compiling bitvec v0.19.6
   Compiling biquad v0.4.2
   Compiling idna v0.2.3
   Compiling regex v1.5.5
   Compiling rand_core v0.6.3
   Compiling cc v1.0.73
   Compiling cstr v0.2.10
   Compiling parking_lot v0.11.2
   Compiling parking_lot v0.12.0
   Compiling crossbeam-queue v0.2.3
   Compiling crossbeam-channel v0.4.4
   Compiling approx v0.5.1
   Compiling num-complex v0.1.43
   Compiling num-traits v0.1.43
   Compiling space v0.17.0
   Compiling topology-traits v0.1.1
   Compiling line_drawing v1.0.0
   Compiling cexpr v0.6.0
   Compiling toml v0.5.9
   Compiling core-graphics-types v0.1.1
   Compiling url v2.2.2
   Compiling hashbrown v0.11.2
   Compiling rand_chacha v0.3.1
   Compiling arrsac v0.10.0
   Compiling rand_xoshiro v0.6.0
   Compiling clang v1.0.3
   Compiling crossbeam-deque v0.8.1
   Compiling ring v0.16.20
   Compiling objc_exception v0.1.2
   Compiling backtrace v0.3.65
   Compiling libmimalloc-sys v0.1.24
   Compiling breakpad-sys v0.1.1
   Compiling crossbeam-deque v0.7.4
   Compiling num-bigint v0.1.44
   Compiling chrono v0.4.19
   Compiling primal-check v0.3.1
   Compiling enum_primitive v0.1.1
   Compiling bitarray v0.9.3
   Compiling enterpolation v0.1.1
   Compiling rand v0.8.5
   Compiling fallible_collections v0.4.4
   Compiling env_logger v0.8.4
   Compiling winres v0.1.12
   Compiling crossbeam v0.7.3
   Compiling opencv-binding-generator v0.41.0
   Compiling num-rational v0.1.42
   Compiling cpp_common v0.5.6
   Compiling synstructure v0.12.6
   Compiling simplelog v0.11.2
   Compiling primal v0.3.0
   Compiling mp4parse v0.12.0
   Compiling thiserror-impl v1.0.30
   Compiling nalgebra-macros v0.1.0
   Compiling derive_more v0.99.17
   Compiling argh_derive v0.1.7
   Compiling prost-derive v0.9.0
   Compiling qmetaobject_impl v0.2.7 (https://github.com/AdrianEddy/qmetaobject-rs.git#769069cc)
   Compiling qutex v0.2.3
   Compiling num v0.1.42
   Compiling cpp_build v0.5.6
   Compiling cpp_macros v0.5.6
   Compiling ffmpeg-sys-next v5.0.1
   Compiling opencv v0.63.1
   Compiling jpeg-decoder v0.1.22
   Compiling thiserror v1.0.30
[... 595 lines cut ...]
cp: cannot overwrite directory /Users/ifeng/gyroflow/target/release/../Frameworks/QtQuick.framework/Headers with non-directory /Users/ifeng/gyroflow/ext/6.3.0/macos/lib/QtQuick.framework/Headers
cp: cannot overwrite directory /Users/ifeng/gyroflow/target/release/../Frameworks/QtQuickControls2.framework/Resources with non-directory /Users/ifeng/gyroflow/ext/6.3.0/macos/lib/QtQuickControls2.framework/Resources
cp: cannot overwrite directory /Users/ifeng/gyroflow/target/release/../Frameworks/QtQuickControls2.framework/Versions/Current with non-directory /Users/ifeng/gyroflow/ext/6.3.0/macos/lib/QtQuickControls2.framework/Versions/Current
cp: cannot overwrite directory /Users/ifeng/gyroflow/target/release/../Frameworks/QtQuickControls2.framework/Headers with non-directory /Users/ifeng/gyroflow/ext/6.3.0/macos/lib/QtQuickControls2.framework/Headers
cp: cannot overwrite directory /Users/ifeng/gyroflow/target/release/../Frameworks/QtQuickControls2Impl.framework/Resources with non-directory /Users/ifeng/gyroflow/ext/6.3.0/macos/lib/QtQuickControls2Impl.framework/Resources
cp: cannot overwrite directory /Users/ifeng/gyroflow/target/release/../Frameworks/QtQuickControls2Impl.framework/Versions/Current with non-directory /Users/ifeng/gyroflow/ext/6.3.0/macos/lib/QtQuickControls2Impl.framework/Versions/Current
cp: cannot overwrite directory /Users/ifeng/gyroflow/target/release/../Frameworks/QtQuickControls2Impl.framework/Headers with non-directory /Users/ifeng/gyroflow/ext/6.3.0/macos/lib/QtQuickControls2Impl.framework/Headers
cp: cannot overwrite directory /Users/ifeng/gyroflow/target/release/../Frameworks/QtQuickDialogs2.framework/Resources with non-directory /Users/ifeng/gyroflow/ext/6.3.0/macos/lib/QtQuickDialogs2.framework/Resources
cp: cannot overwrite directory /Users/ifeng/gyroflow/target/release/../Frameworks/QtQuickDialogs2.framework/Versions/Current with non-directory /Users/ifeng/gyroflow/ext/6.3.0/macos/lib/QtQuickDialogs2.framework/Versions/Current
cp: cannot overwrite directory /Users/ifeng/gyroflow/target/release/../Frameworks/QtQuickDialogs2.framework/Headers with non-directory /Users/ifeng/gyroflow/ext/6.3.0/macos/lib/QtQuickDialogs2.framework/Headers
cp: cannot overwrite directory /Users/ifeng/gyroflow/target/release/../Frameworks/QtQuickDialogs2QuickImpl.framework/Resources with non-directory /Users/ifeng/gyroflow/ext/6.3.0/macos/lib/QtQuickDialogs2QuickImpl.framework/Resources
cp: cannot overwrite directory /Users/ifeng/gyroflow/target/release/../Frameworks/QtQuickDialogs2QuickImpl.framework/Versions/Current with non-directory /Users/ifeng/gyroflow/ext/6.3.0/macos/lib/QtQuickDialogs2QuickImpl.framework/Versions/Current
cp: cannot overwrite directory /Users/ifeng/gyroflow/target/release/../Frameworks/QtQuickDialogs2QuickImpl.framework/Headers with non-directory /Users/ifeng/gyroflow/ext/6.3.0/macos/lib/QtQuickDialogs2QuickImpl.framework/Headers
cp: cannot overwrite directory /Users/ifeng/gyroflow/target/release/../Frameworks/QtQuickDialogs2Utils.framework/Resources with non-directory /Users/ifeng/gyroflow/ext/6.3.0/macos/lib/QtQuickDialogs2Utils.framework/Resources
[... cut ...]
cp: cannot overwrite directory /Users/ifeng/gyroflow/target/release/../Frameworks/QtWidgets.framework/Resources with non-directory /Users/ifeng/gyroflow/ext/6.3.0/macos/lib/QtWidgets.framework/Resources
cp: cannot overwrite directory /Users/ifeng/gyroflow/target/release/../Frameworks/QtWidgets.framework/Versions/Current with non-directory /Users/ifeng/gyroflow/ext/6.3.0/macos/lib/QtWidgets.framework/Versions/Current
cp: cannot overwrite directory /Users/ifeng/gyroflow/target/release/../Frameworks/QtWidgets.framework/Headers with non-directory /Users/ifeng/gyroflow/ext/6.3.0/macos/lib/QtWidgets.framework/Headers

AdrianEddy, 2022-04-26:
follow the building steps carefully, you're probably missing `export DYLD_FALLBACK_LIBRARY_PATH="$(xcode-select --print-path)/Toolchains/XcodeDefault.xctoolchain/usr/lib/"`

kxifenger, 2022-04-27:
> follow the building steps carefully, you're probably missing `export DYLD_FALLBACK_LIBRARY_PATH="$(xcode-select --print-path)/Toolchains/XcodeDefault.xctoolchain/usr/lib/"`

thank you i am sure that i run this command `export DYLD_FALLBACK_LIBRARY_PATH="$(xcode-select --print-path)/Toolchains/XcodeDefault.xctoolchain/usr/lib/"`

kxifenger, 2022-04-27:
> thank you i am sure that i run this command `export DYLD_FALLBACK_LIBRARY_PATH="$(xcode-select --print-path)/Toolchains/XcodeDefault.xctoolchain/usr/lib/"`

thank you 
i run this command
export DYLD_FALLBACK_LIBRARY_PATH="$(xcode-select --print-path)/Toolchains/XcodeDefault.xctoolchain/usr/lib/" 
and the DYLD_FALLBACK_LIBRARY_PATH is /Library/Developer/CommandLineTools/Toolchains/XcodeDefault.xctoolchain/usr/lib/
but i do not have this path in my mac.
so i use this command export DYLD_FALLBACK_LIBRARY_PATH="/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/" and i can solve this problem.

kxifenger, 2022-04-27:
> follow the building steps carefully, you're probably missing `export DYLD_FALLBACK_LIBRARY_PATH="$(xcode-select --print-path)/Toolchains/XcodeDefault.xctoolchain/usr/lib/"`

thank you
i run this command
export DYLD_FALLBACK_LIBRARY_PATH="$(xcode-select --print-path)/Toolchains/XcodeDefault.xctoolchain/usr/lib/"
and the DYLD_FALLBACK_LIBRARY_PATH is /Library/Developer/CommandLineTools/Toolchains/XcodeDefault.xctoolchain/usr/lib/
but i do not have this path in my mac.
so i use this command export DYLD_FALLBACK_LIBRARY_PATH="/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/" and i can solve this problem.

kxifenger, 2022-04-27:
there is new problem  when i run cargo run --release   
i do not know how to solve it.

Compiling gyroflow v1.0.0-rc5 (/Users/ifeng/gyroflow)
error: linking with `cc` failed: exit status: 1
  |
  = note: "cc" "-m64" "-arch" "x86_64" "/Users/ifeng/gyroflow/target/release/deps/gyroflow-948e33ac5ae311c2.gyroflow.7e75a051-cgu.0.rcgu.o" "/Users/ifeng/gyroflow/target/release/deps/gyroflow-948e33ac5ae311c2.gyroflow.7e75a051-cgu.1.rcgu.o" "/Users/ifeng/gyroflow/target/release/deps/gyroflow-948e33ac5ae311c2.gyroflow.7e75a051-cgu.10.rcgu.o" "/Users/ifeng/gyroflow/target/release/deps/gyroflow-948e33ac5ae311c2.gyroflow.7e75a051-cgu.11.rcgu.o" "/Users/ifeng/gyroflow/target/release/deps/gyroflow-948e33ac5ae311c2.gyroflow.7e75a051-cgu.12.rcgu.o" "/Users/ifeng/gyroflow/target/release/deps/gyroflow-948e33ac5ae311c2.gyroflow.7e75a051-cgu.13.rcgu.o" "/Users/ifeng/gyroflow/target/release/deps/gyroflow-948e33ac5ae311c2.gyroflow.7e75a051-cgu.14.rcgu.o" "/Users/ifeng/gyroflow/target/release/deps/gyroflow-948e33ac5ae311c2.gyroflow.7e75a051-cgu.15.rcgu.o" "/Users/ifeng/gyroflow/target/release/deps/gyroflow-948e33ac5ae311c2.gyroflow.7e75a051-cgu.2.rcgu.o" "/Users/ifeng/gyroflow/target/release/deps/gyroflow-948e33ac5ae311c2.gyroflow.7e75a051-cgu.3.rcgu.o" "/Users/ifeng/gyroflow/target/release/deps/gyroflow-948e33ac5ae311c2.gyroflow.7e75a051-cgu.4.rcgu.o" "/Users/ifeng/gyroflow/target/release/deps/gyroflow-948e33ac5ae311c2.gyroflow.7e75a051-cgu.5.rcgu.o" "/Users/ifeng/gyroflow/target/release/deps/gyroflow-948e33ac5ae311c2.gyroflow.7e75a051-cgu.6.rcgu.o" "/Users/ifeng/gyroflow/target/release/deps/gyroflow-948e33ac5ae311c2.gyroflow.7e75a051-cgu.7.rcgu.o" "/Users/ifeng/gyroflow/target/release/deps/gyroflow-948e33ac5ae311c2.gyroflow.7e75a051-cgu.8.rcgu.o" "/Users/ifeng/gyroflow/target/release/deps/gyroflow-948e33ac5ae311c2.gyroflow.7e75a051-cgu.9.rcgu.o" "/Users/ifeng/gyroflow/target/release/deps/gyroflow-948e33ac5ae311c2.46126hzjh7cqwjbz.rcgu.o" "-L" "/Users/ifeng/gyroflow/target/release/deps" "-L" "/Users/ifeng/gyroflow/ext/ffmpeg-x86_64/lib" "-L" "/Users/ifeng/gyroflow/target/release/build/gyroflow-6190eef101f68439/out" "-L" "/Users/ifeng/gyroflow/target/release/build/breakpad-sys-f3181e00bd53a932/out" "-L" "/Users/ifeng/gyroflow/ext/ffmpeg-x86_64/lib" "-L" "/Users/ifeng/gyroflow/target/release/build/libmimalloc-sys-3f736a33a5a844c6/out" "-L" "/Users/ifeng/gyroflow/target/release/build/opencv-d7f980ae7a614466/out" "-L" "/Users/ifeng/gyroflow/ext/vcpkg/installed/x64-osx-release/lib" "-F" "/Users/ifeng/gyroflow/ext/vcpkg/installed/x64-osx-release/lib" "-L" "/Users/ifeng/gyroflow/target/release/build/objc_exception-f1fe6427708c669b/out" "-L" "/Users/ifeng/gyroflow/target/release/build/qmetaobject-d1d788fbfb1ca9fd/out" "-L" "/Users/ifeng/gyroflow/target/release/build/qttypes-b118f8ace8036dcc/out" "-F" "/Users/ifeng/gyroflow/ext/6.3.0/macos/lib" "-F" "/Users/ifeng/gyroflow/target/release/build/qml-video-rs-33050814ac2366c9/out/mdk-sdk/lib/" "-L" "/Users/ifeng/gyroflow/target/release/build/qml-video-rs-33050814ac2366c9/out" "-L" "/Users/ifeng/gyroflow/target/release/build/ring-c874327af267dd34/out" "-L" "/Users/ifeng/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/x86_64-apple-darwin/lib" "-lz" "-lbz2" "-lxml2" "-framework" "AudioToolbox" "-framework" "VideoToolbox" "-framework" "QuartzCore" "-framework" "Foundation" "-framework" "CoreMedia" "-framework" "CoreAudio" "-framework" "CoreVideo" "-framework" "CoreServices" "-framework" "CoreFoundation" "-framework" "AppKit" "-framework" "OpenGL" "-Wl,-force_load" "-Wl,/Users/ifeng/gyroflow/ext/ffmpeg-x86_64/lib/libx264.a" "-Wl,-force_load" "-Wl,/Users/ifeng/gyroflow/ext/ffmpeg-x86_64/lib/libx265.a" "-Wl,-force_load" "-Wl,/Users/ifeng/gyroflow/target/release/build/gyroflow-6190eef101f68439/out/librust_cpp_generated.a" "-lc++" "/Users/ifeng/gyroflow/target/release/deps/libkeep_awake-273cad767ad22978.rlib" "/Users/ifeng/gyroflow/target/release/deps/libmach-ef581bd85f067d62.rlib" "/Users/ifeng/gyroflow/target/release/deps/libwhoami-75e630cf4fc37cbe.rlib" "/Users/ifeng/gyroflow/target/release/deps/libsemver-be1cfd8f6f465926.rlib" "/Users/ifeng/gyroflow/target/release/deps/libureq-390895292dddf083.rlib" "/Users/ifeng/gyroflow/target/release/deps/libwebpki_roots-89bb9b801b448331.rlib" "/Users/ifeng/gyroflow/target/release/deps/libbase64-d4085fe3c72f23e8.rlib" "/Users/ifeng/gyroflow/target/release/deps/librustls-c684c35ce7601e04.rlib" "/Users/ifeng/gyroflow/target/release/deps/libsct-7986ffd3d596cdf4.rlib" "/Users/ifeng/gyroflow/target/release/deps/libwebpki-4342c68c4e531b3f.rlib" "/Users/ifeng/gyroflow/target/release/deps/libring-2e8aabc1c11f70e3.rlib" "/Users/ifeng/gyroflow/target/release/deps/libspin-6e2f160f45b8b244.rlib" "/Users/ifeng/gyroflow/target/release/deps/libuntrusted-01a75cb97877304c.rlib" "/Users/ifeng/gyroflow/target/release/deps/libchunked_transfer-48b78effdf2a9151.rlib" "/Users/ifeng/gyroflow/target/release/deps/liburl-0e11d65f3dea6ab7.rlib" "/Users/ifeng/gyroflow/target/release/deps/libidna-fff2728436d3d75b.rlib" "/Users/ifeng/gyroflow/target/release/deps/libunicode_normalization-1446dce2e4354f8c.rlib" "/Users/ifeng/gyroflow/target/release/deps/libtinyvec-0319fc182a80e531.rlib" "/Users/ifeng/gyroflow/target/release/deps/libtinyvec_macros-eb1f5a1f240284ed.rlib" "/Users/ifeng/gyroflow/target/release/deps/libunicode_bidi-73d3d9427fdc843f.rlib" "/Users/ifeng/gyroflow/target/release/deps/libform_urlencoded-084b1771d83956ef.rlib" "/Users/ifeng/gyroflow/target/release/deps [... cut ...] ickControls2" "-lSystem" "-lresolv" "-lc" "-lm" "-liconv" "-L" "/Users/ifeng/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/x86_64-apple-darwin/lib" "-o" "/Users/ifeng/gyroflow/target/release/deps/gyroflow-948e33ac5ae311c2" "-Wl,-dead_strip" "-Wl,-rpath,@loader_path/../Frameworks"
  = note: ld: library not found for -ltegra_hal
          clang: error: linker command failed with exit code 1 (use -v to see invocation)
          

error: could not compile `gyroflow` due to previous error

closed 2022-04-28 by AdrianEddy

kxifenger, 2022-04-29:
> `= note: ld: library not found for -ltegra_hal` this happens when you try to use x64 libraries when building for arm64 - the solution is to replace `x86_64` to `arm64` and `x64-osx-release` to `arm64-osx` in __env-macos.sh This step is mentioned in the building instructions already

thank you
i am confused that you said you mentioned in the building instructions already.
i read the building instructions carefully. my mac is not the Apple M1 chip. so i do not need to 
For building on Apple M1, change ffmpeg-x64_64 -> ffmpeg-arm64 and x64-osx-release -> arm64-osx in __env-macos.sh

AdrianEddy, 2022-04-29:
Hmm, so which Mac do you have? I had this exact error when trying to build on M1 mac without changing the paths.
Did you manage to build Gyroflow finally?

kxifenger, 2022-04-30:
> Hmm, so which Mac do you have? I had this exact error when trying to build on M1 mac without changing the paths. Did you manage to build Gyroflow finally?

my mac is mac os monterey 12.1 and Intel Core i7 . now i still can not build Gyroflow .

kxifenger, 2022-04-30:
i try to remove libtegra_hal in  OPENCV_LINK_LIBS="opencv_core,opencv_calib3d,opencv_features2d,opencv_imgproc,opencv_video,opencv_flann,libtegra_hal".
and i got this error
ifeng@ifengdeMacBook-Pro gyroflow % cargo run --release   
   Compiling opencv v0.63.1
   Compiling gyroflow v1.0.0-rc5 (/Users/ifeng/gyroflow)
   Compiling gyroflow-core v1.0.0-rc5 (/Users/ifeng/gyroflow/src/core)
    Finished release [optimized] target(s) in 1m 32s
     Running `target/release/gyroflow`
dyld[1864]: Library not loaded: @rpath/QtCore.framework/Versions/A/QtCore
  Referenced from: /Users/ifeng/gyroflow/target/release/gyroflow
  Reason: tried: '/Users/ifeng/gyroflow/target/release/../Frameworks/QtCore.framework/Versions/A/QtCore' (no such file), '/Users/ifeng/gyroflow/target/release/../Frameworks/QtCore.framework/Versions/A/QtCore' (no such file), '/Library/Frameworks/QtCore.framework/Versions/A/QtCore' (no such file), '/System/Library/Frameworks/QtCore.framework/Versions/A/QtCore' (no such file)
zsh: abort      cargo run --release

AdrianEddy, 2022-04-30:
This actually built Gyroflow correctly, now just follow last step from instructions (run deploy-macos once and then cargo run again)

kxifenger, 2022-04-30:
thank you i can build Gyroflow now on my mac. 
i wonder that there is something wrong that i remove libtegra_hal in OPENCV_LINK_LIBS="opencv_core,opencv_calib3d,opencv_features2d,opencv_imgproc,opencv_video,opencv_flann,libtegra_hal".
if i do not remove it, i can not build Gyroflow due to = note: ld: library not found for -ltegra_hal

AdrianEddy, 2022-04-30:
you can safely remove libtegra_hal and everything will still work, Gyroflow doesn't use it directly, it's a dependency of opencv.