gyroflow/gyroflow #319: how to debug this project

kxifenger, 2022-04-27:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Now i can run this project in terminal. 
i want to know there is any ways to debug in this project.
for example if i want to set breakpoint in code，what should i do.

closed 2022-04-29 by kxifenger

AdrianEddy, 2022-04-29:
Did you manage to debug it? I was going to check how to do this on macos, but your best bet should be to use Attach to Process and C++ extension for VS Code (cppvsdbg)

kxifenger, 2022-04-30:
i still do not know how to debug it? if i want to debug in windows, what should i do.
and i want to know How do you do it.

AdrianEddy, 2022-04-30:
On Windows: Install C++ extension for vscode, run the project from command line (without `--release` so just `cargo run`) and then in VS Code use "Attach to process". On MacOS should be similar but i didn't test