gyroflow/gyroflow #320: Memory leak in Auto sync

dncoder, 2022-05-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.0.0-rc5

### What operating system are you using?

Windows 10 build 19043

### What GPU are you using?

Nvidia GeForce GTX 1060

### What happened?

The memory is not freed after the autosync process is done. Every time the auto sync is performed on any point, the process takes up nearly another 1 GB of RAM.
This always happens in the following conditions:
- Single point "Autosync here" or "Autosync" button with multiple points
- Optical flow method: OpenCV or AKAZE
- Use GPU decoding enabled or disabled

If you save the project, restart the program and open it, it's fine again.
The latest release 1.0.0 rc5 has a limit of maximum 10 points for autosync. I've also tried the older release 1.0.0 rc3 which has no such limit. After about 2 minutes of analysis with 40 sync points, the process reaches about 25+ GB of RAM usage and crashes on my 32 GB machine. 
Can you confirm such issue or should I try this on another system? Thank you.
[image]

### Relevant log output

```shell
18:00:48 [INFO] Lens profiles directory: "\\\\?\\D:\\soft\\Gyroflow-windows64\\camera_presets"                          18:00:48 [INFO] Loaded 1290 lens profiles in 15.296ms                                                                   18:00:48 [INFO] OpenCL Platform: NVIDIA CUDA, Device: NVIDIA Corporation NVIDIA GeForce GTX 1060 6GB                    18:00:48 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA                                       18:00:48 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA                                   [src\rendering\mod.rs:45] gpu_type = NVIDIA                                                                             18:00:51 [DEBUG] (14) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""                                           18:00:51 [DEBUG] (15) MDKPlayer: [in m_player->onEvent] "thread.video" ""                                               18:00:51 [DEBUG] (16) MDKPlayer: [in m_player->onEvent] "thread.audio" ""                                               18:00:51 [DEBUG] (16) MDKPlayer: [in m_player->onEvent] "decoder.audio" "FFmpeg"                                        18:00:51 [DEBUG] (16) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""                                           18:00:51 [DEBUG] (17) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Direct3D11              18:00:51 [DEBUG] (17) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x20cc9cbb4f0) QSize(1373, 772)   18:00:51 [DEBUG] (15) MDKPlayer: [in m_player->onEvent] "decoder.video" "AVCodec.D3D11"                                 18:00:52 [DEBUG] (17) MDKPlayer: [in m_player->onEvent] "render.video" "1st_frame"                                      18:00:52 [INFO] surface size: 2704x1520                                                                                 18:00:52 [DEBUG] (1) qml: Setting param time_constant 3                                                                 18:00:52 [DEBUG] (1) qml: Setting param time_constant2 0.1                                                              18:00:52 [DEBUG] (1) qml: Setting param velocity_factor 0.3                                                             18:00:52 [DEBUG] (17) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Direct3D11              18:00:52 [DEBUG] (17) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x20ccb061d40) QSize(2704, 1520)  18:00:55 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [dec] codec type AV_HWDEVICE_TYPE_DXVA2 0                          18:00:55 [DEBUG] (2) gyroflow::rendering: Selected HW backend AV_HWDEVICE_TYPE_DXVA2 (dxva2) with format Some(AV_PIX_FMT_DXVA2_VLD)                                                                                                                                                                                                                                     18:01:19 [INFO] Setting offset at 55353.1592: 21327.6800 (cost 20543.1390)                                              18:01:19 [INFO] Setting offset at 166092.9238: 21730.6600 (cost 4205.2698)                                              18:01:19 [INFO] Setting offset at 276815.9652: 22219.9900 (cost 25974.8785)                                             18:01:19 [INFO] Setting offset at 387555.7298: 22726.4700 (cost 3121.2726)
```

AdrianEddy, 2022-05-01:
Does the memory get cleared after loading a different file without program restart?

AdrianEddy, 2022-05-01:
Did you also try latest dev build? https://gyroflow.xyz/devbuild/?autodownload

AdrianEddy, 2022-05-01:
It's actually not a memory leak, but the frames used to analyze optical flow are held in memory with the optical flow and they get cleared after loading another file. This is perhaps unnecessary, I'll look into it

dncoder, 2022-05-01:
Still same exact behavior on the latest dev build 2022-05-01. The memory doesn't get cleared after loading a different file, even multiple times. Loading a new file itself doesn't produce any growth in memory, only autosync does. Each time it takes away about ~1500MB of RAM with 2.7k video file and ~400MB with 1080p.  The gyro data is Betaflight CSV.

AdrianEddy, 2022-05-03:
There are multiple reasons for that, I optimized some in 1fb4dff1fc9d325f2311941ea1ecc1b03b354c1b, you can check out latest dev build https://gyroflow.xyz/devbuild/?autodownload
- Every frame used for analysis was kept in the autosync structure - now it is cleared after syncing
- Autosync algorithms apparently allocate a lot of thread-local variables and we're using thread pool - that thread-local data was never cleared, because threads were not destructed - solved that by creating a dedicated thread-pool for autosync
- A lot of that memory is memory kept by the custom allocator - we use [MiMalloc](https://github.com/microsoft/mimalloc) - it's not leaked memory, but it'll be reused by the program later, or returned to the system after some time
- Image data was moved multiple times, which in Rust means copying the memory to a different location, which in conjunction with MiMalloc left that memory reserved for the program instead of returning to the system - solved this by using a reference to a single image-data location in memory

You can try latest dev build and let me know how much that helped

closed 2022-05-03 by AdrianEddy

dncoder, 2022-05-03:
Yes, indeed it handles memory so much better now!
You can see it by the test I did, waited for about a minute after each step

before 1fb4dff :
```
+---+----------------------------------+-------------------------------------+
| # |               Step               |              RAM Usage              |
+---+----------------------------------+-------------------------------------+
| 1 | Open Gyroflow v1.0.0-rc5 (gh465) | 268 MB                              |
| 2 | Load project                     | 1340 Mz                             |
| 3 | Run autosync with 10 points      | 15860 MB                            |
| 4 | Run autosync with 9 points       | 27706 MB (!)                        |
| 5 | Run autosync with 8 points       | crashed at about 97% with 28000+ MB |
+---+----------------------------------+-------------------------------------+
```
after 1fb4dff :
```
+---+----------------------------------+-----------+
| # |               Step               | RAM Usage |
+---+----------------------------------+-----------+
| 1 | Open Gyroflow v1.0.0-rc5 (gh478) | 270 MB    |
| 2 | Load project                     | 1343 MB   |
| 4 | Run autosync with 10 points      | 8672 MB   |
| 5 | Run autosync with 9 points       | 8656 MB   |
| 6 | Run autosync with 8 points       | 8266 MB   |
| 7 | Run autosync with 7 points       | 7724 MB   |
+---+----------------------------------+-----------+
```

So after 34 points it still doesn't use too much and keeps going down with each new autosync point added, staying at about 7 GB.
Thank you so much for this awesome work!