gyroflow/gyroflow #323: rendering % not shows up correctly in taskbar

WizzX-FPV, 2022-05-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.0.0-RC5 (gh465)

### What operating system are you using?

Windows 10

### What GPU are you using?

Intel UHD 620

### What happened?

During rendering I can see correct percentage of how much of video is already rendered on main screen.
But Gyroflow icon in the taskbar looks like percentage value is different as on main window (I suppose it should be the same).
[image]

AdrianEddy, 2022-05-02:
And what items do you have in the render queue?

WizzX-FPV, 2022-05-02:
not using render queue, only single file

AdrianEddy, 2022-05-02:
Yes but even single file is added to the render queue, it's probably showing progress of the entire render queue (which includes already rendered items)

closed 2022-05-03 by AdrianEddy

WizzX-FPV, 2022-05-04:
I can confirm that with last automated build it works OK, thank you Adrian, good job!