gyroflow/gyroflow #325: Video will not load

bigjoncoop, 2022-05-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.0 rc5

### What operating system are you using?

Windows 10 Pro

### What GPU are you using?

AMD RYZEN integrated

### What happened?

Starting today I am unable to load a video clip into GYROFLOW...

After selecting a video the status Circle just keeps spinning. I've tried multiple different videos and different formats.

All the videos I've tried open fine in Windows Media Player and VLC

GYROFLOW  was working fine up until now
I also tried deleting the GYROFLOW folder and "reinstalling"

AdrianEddy, 2022-05-02:
Please download latest dev build and attach gyroflow.log file https://gyroflow.xyz/devbuild/?autodownload

bigjoncoop, 2022-05-04:
@AdrianEddy 

Here is the gyroflow log output:
(If I just let GyroFlow run and keep trying to load the file the line "[DEBUG] (12) mdk: [MDK] seeking media's position is -16=====" will just keep going and make the file EXTREMELY long. So for this log I just loaded the video file and closed GyroFlow after 10sec or so....

[gyroflow.log](https://github.com/gyroflow/gyroflow/files/8617992/gyroflow.log)

AdrianEddy, 2022-05-04:
Does it happen with all videos, or just this one? Can you send me that video file?

AdrianEddy, 2022-05-04:
Also install all drivers updates and Windows updates, it happened before that an Windows update broke loading videos

bigjoncoop, 2022-05-04:
@AdrianEddy

Q. Does it happen with all videos, or just this one? Can you send me that video file?

A. I've tried numerous different videos. Same issue with all of them. I'm not sure if they were all MP4 or not. But they were definitely not all from the DJI Goggles. They were just random videos.
( I will try to load some videos of different formats in a few minutes and let you know what happens)

Q. Also install all drivers updates and Windows updates, it happened before that an Windows update broke loading videos

A.  I have Windows updates permanently disabled. 100% positive there has been no updates since the last time GYROFLOW had worked until now.

The only thing I can think of that has change is oh, I had removed my GTX 1650 and replaced it with a cheap a AMD 5450... Which is running the newest drivers available. ( I will run the Nvidea driver uninstaller program and reinstall the AMD drivers and see if that changes everything)

I will let you know what happens shortly

AdrianEddy, 2022-05-04:
Do you also have Microsoft Store blocked? eg. Microsoft HEVC Extensions is updated through Microsoft Store and it caused problems before.
You can also try adding `set MDK_DECODERS=FFmpeg` as a second line in `Gyroflow_with_console.bat` and running it

bigjoncoop, 2022-05-08:
@AdrianEddy

No the Microsoft store is still functional.

I will try to "set MDK_DECODERS=FFmpeg as a second line in Gyroflow_with_console.bat" and let you know what happens shortly. 

* also I was thinking that the only thing that has changed since the last time that GYROFLOW worked is I had removed my GTX 1650 and installed a cheap AMD 5450 gpu. So last night I ran the DDU driver uninstaller and completely removed all of the NVIDIA drivers and reinstalled the AMD drivers.

But unfortunately that did not help either

p-masek, 2022-05-19:
I had exactly the same issue (same error in log, GPU on 80 % during import). 
As per above I thought it is because of Windows updates as it also worked fine couple days ago. But then I have realized I have changed 3D setting in Nvidia control panel. When I return it back to 'Let the 3D application decide" (from my preference settings 'Performance') it has started to work. Maybe on AMD is something similar.

[image]

AdrianEddy, 2022-05-19:
How about this:
Add `set NO_OPENCL=1` as a second line in `Gyroflow_with_console.bat` and run it (just add the line, don't replace or remove other lines)

ericogr, 2022-05-22:
I'm using linux ubuntu 20.04, gyroflow 1.0.1 and nvidia gtx 1070ti.
I can't open some videos (not all) from Runcam Thumb (firmware 2.1.0) with or without gyro data.

From logs, I got (repeated times):
```
19:24:13 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "decoder.audio" "FFmpeg"
19:24:13 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "decoder.audio" "FFmpeg"
19:24:13 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "decoder.audio" "FFmpeg"
```

and the video won't load. But I can play it from VLC.

enryco99, 2022-06-06:
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/8844284/gyroflow.log)
I've a similar problem. First video that i tried to import took a while and the program look stuck also after few minute so I've closed that. Now every other video that I try to import cause a crash.

closed 2022-06-20 by AdrianEddy