gyroflow/gyroflow #329: Auto lens calibration is very slow and often seems to get completely stuck (but eventually finishes)

avsaase, 2022-05-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

gh481

### What operating system are you using?

Windows 10

### What GPU are you using?

AMD Radeon RX 570

### What happened?

Basically the title. It starts quite fast and the estimated time remaining is 30 seconds but after ~25% it completely grinds to a halt. After a minute or so the progress bar gets moving again, only to stop again later. Eventually, it finishes but the process can take almost 10 minutes. When it halts the frame rate and estimated time remaining don't get updated anymore and thus become very inaccurate. I guess it's not a huge problem because I am able to make a lens profile but I recall this being _much_ faster in the past.

Calibration video and log file are available [here](https://mega.nz/folder/x0NyxCTZ#1sc8udaFXz43tsoySeBmMw).

Probably unrelated and really a minor problem but in the past when exporting a newly created lens profile, the Windows save as window had a suggested file name with the camera name and settings. Now the file name field is just empty: 
[image]

Captain0Kangaroo, 2022-05-09:
I'm finding the same when make a lens profile. Its taking about 1h to complete.

avsaase, 2022-06-08:
Cool! I'll try out the new pattern tomorrow.