gyroflow/gyroflow #330: Adding calibration points manually in Lens calibrator results in a degenerate solution

VladimirP1, 2022-05-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.0.0

### What operating system are you using?

Windows 11

### What happened?

Steps to reproduce:
* Load a chessboard video into the Lens calibrator
* Try to calibrate the lens by adding points manually

Result:
* The undistorted image degenerates to a single pixel

VladimirP1, 2022-05-05:
Bisected to d6c6dcde4387b958c77d3f4752fe604ac61e6a66

AdrianEddy, 2022-05-05:
Fixed in 1ff396ee2cd980cda70fb925fc4fb52e7f197dea