gyroflow/gyroflow #331: Lens Profile dropdown immediately dismisses while typing

Stampede10343, 2022-05-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.0.0 AppImage

### What operating system are you using?

Ubuntu 20, 

### What GPU are you using?

Nvidia GTX2070

### What happened?

When searching for lens profiles, the recommendation drop down just flashes as you type and does not come up or show any results.

I don't see anything

### Relevant log output

```shell
MDK 0.14.2 (git 9bd409e) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang14.0.1 GCC4.2.1; libc++14000; 13:08:48 May  3 2022
17:59:57 [WARN] Binding on animation is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
17:59:57 [WARN] Binding on indicator is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
17:59:57 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
17:59:57 [WARN] Binding on popup is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
17:59:57 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
17:59:58 [INFO] Lens profiles directory: "/tmp/.mount_GyrofloQn8oF/camera_presets"
17:59:59 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("NVIDIA CUDA") ["NVIDIA GeForce RTX 2070"]
17:59:59 [INFO] OpenCL Platform: NVIDIA CUDA, Device: NVIDIA Corporation NVIDIA GeForce RTX 2070
17:59:59 [INFO] Loaded 3025 lens profiles in 87.611ms
17:59:59 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
17:59:59 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
[src/rendering/mod.rs:50] gpu_type = NVIDIA
17:59:59 [INFO] Latest version: v1.0.0, current version: 1.0.0 (gh493)
18:01:37 [DEBUG] (10) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
18:01:37 [DEBUG] (11) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
18:01:37 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "thread.video" ""
18:01:37 [DEBUG] (11) MDKPlayer: [in m_player->onEvent] "decoder.audio" "FFmpeg"
18:01:37 [DEBUG] (11) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
18:01:37 [DEBUG] (13) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::OpenGL
18:01:37 [DEBUG] (13) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x7fcfd65a5ef0) QSize(1031, 580)
18:01:37 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "decoder.video" "AVCodec.VDPAU"
get Context 0x7fcfd45bf610/0x7fcfd40060a8 has no local data for key 0x2, create 1(0x7fcfd45220d0)
get Context 0x7fcfd45bf610/0x7fcfd40060a8 has no local data for key 0x1, create 1(0x7fcfd4678b30)
get Context 0x7fcfd45bf610/0x7fcfd40060a8 has no local data for key (nil), create 1(0x7fcfd65c1700)
18:01:38 [DEBUG] (13) MDKPlayer: [in m_player->onEvent] "render.video" "1st_frame"
18:01:38 [INFO] surface size: 1280x720
18:01:38 [DEBUG] (13) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::OpenGL
18:01:38 [DEBUG] (13) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x7fcfd6aad6f0) QSize(1280, 720)
```

AdrianEddy, 2022-05-05:
🤔 I swear this was working just a few days ago

Sinewavz, 2022-05-05:
Same issue here.

AdrianEddy, 2022-05-05:
Fixed in 1.0.1 https://github.com/gyroflow/gyroflow/releases/tag/v1.0.1

closed 2022-05-05 by AdrianEddy

Stampede10343, 2022-05-05:
Sweet, confirmed its working for me now! Thanks for the quick fix