gyroflow/gyroflow #336: GPU Encoding

billjow, 2022-05-07:
Hi, only recently started using Gyroflow, with 1.0.0-rc5, with no problems including being able to use GPU encoding. Upgraded to 1.0.0, but it could not detect my GPU, and nor could 1.0.1. Am using a Lenovo Yoga 510-141SK laptop - seems to be using Intel HD Graphics.

AdrianEddy, 2022-05-07:
Include gyroflow.log file

billjow, 2022-05-08:
Hi Adrian,

 

Please find attached the log file from 1.0.1 and the log file from 1.0.0.

 

Please also find attached a screenshot of the error message from 1.0.0.

 

Kind Regards,

 

Billy

 

Bill Jowitt

Tel: 07790 491385

AdrianEddy, 2022-05-09:
It didn't attach the files, please reply on github - email attachments are not included

billjow, 2022-05-09:
Sorry about that.
[gyroflowlogs.zip](https://github.com/gyroflow/gyroflow/files/8649792/gyroflowlogs.zip)

Logs and jpg of error message attached in the zip file.

AdrianEddy, 2022-05-09:
hmm, are you sure it worked in rc5? Can you also send me log from rc5 after rendering?

billjow, 2022-05-09:
On a call at the moment, will have to render another file as I opened the version up this morning to check which version was in which folder. Will get back to you in about an hour. Cheers, Bill

billjow, 2022-05-09:
[gyroflow-100-rc5.zip](https://github.com/gyroflow/gyroflow/files/8650354/gyroflow-100-rc5.zip)

log file for v100-rc5 in the attached. Cheers, Bill

AdrianEddy, 2022-05-09:
ok looks like rc5 detected your Intel GPU, and v1.0.1 detects your AMD GPU. Do you have latest drivers installed for AMD?

billjow, 2022-05-09:
[Gyroflow log error and graphics drivers status.zip](https://github.com/gyroflow/gyroflow/files/8650992/Gyroflow.log.error.and.graphics.drivers.status.zip)

I went through a fairly tortuous process but updated the AMD driver - I was not able to update the Intel driver, see graphic in zip file, I went through a Lenovo update process and there were no drivers to update. v1.0.1 still not working, see the log file and the error message in the zip. Thanks for your patience, Bill

billjow, 2022-05-09:
1.0.0-rc5 still working
[gyroflow.zip](https://github.com/gyroflow/gyroflow/files/8651040/gyroflow.zip)
.

AdrianEddy, 2022-06-13:
Please try latest dev build, and set `Device for rendering` to your Intel GPU in Export -> Advanced
https://gyroflow.xyz/devbuild/?autodownload
Let me know if that works

billjow, 2022-06-15:
Hi Adrian,

 

It works well, rendered at around 18fps which is 50% more than before on 1.0.0 – thanks!

 

Kind Regards,

 

Bill

 

Bill Jowitt

Tel: 07790 491385

 

https://gyroflow.xyz/devbuild/?autodownload

billjow, 2022-06-15:
Hi Adrian,

 

Tried to use v1.1.0, but it did not work, even though I selected the same options. Screenshot attached.

 

Kind Regards,

 

Bill

 

Bill Jowitt

Tel: 07790 491385

 

 

 

 

 

 

 

https://gyroflow.xyz/devbuild/?autodownload

AdrianEddy, 2022-06-15:
Hmm, nothing was changed in this regard between dev version and v1.1, can you please double check again?

billjow, 2022-06-16:
Hi Adrian,

 

Tested it again on a short video, identical settings on both versions. Same result. Log files attached – let me know if there is anything else you need.

 

Kind Regards,

 

Billy

 

Bill Jowitt

Tel: 07790 491385

AdrianEddy, 2022-06-16:
Please attach the logs on Github - sending attachments in email reply doesn't work

billjow, 2022-06-16:
[gyroflow_devbuild.zip](https://github.com/gyroflow/gyroflow/files/8920042/gyroflow_devbuild.zip)
[gyroflow_1.1.0.zip](https://github.com/gyroflow/gyroflow/files/8920046/gyroflow_1.1.0.zip)

Attached as requested, as .zips because GitHub would not accept .logs.

AdrianEddy, 2022-06-16:
and which device you selected in Export settings->Advanced? can you show me a screenshot of the list and the one you selected? In v1.1.0

billjow, 2022-06-17:
[image]
[image]
[image]
[image]

Screenshots attached showing the lists for encoding and decoding devices and error messages with and without GPU decoding selected.

AdrianEddy, 2022-06-17:
Thanks, I think I found the problem, please test with gh574 version: https://gyroflow.xyz/devbuild/?autodownload

AdrianEddy, 2022-06-17:
> That version works, thanks. Just an observation - it seems to run slower than the previous dev version, around 9-10 fps.

You can set Intel for rendering, but AMD for processing, that should be faster

billjow, 2022-06-18:
Thanks for that - on a short clip it got up to 16.6fps - terrific!