gyroflow/gyroflow #338: MP4 Video FC Blackbox log issue

BludzUK, 2022-05-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.0.1

### What operating system are you using?

Windows 10

### What GPU are you using?

3090

### What happened?

I'm trying the new version and when I add blackbox log for gyro data, the video disappears. When I play you I see occasional flashes but not frames of the video. I have tried with files used on previous version and they do the same. The video is an MP4, hope that helps. Let me know what else you need to know. Thanks in advance

AdrianEddy, 2022-05-08:
Upload these files: video, blackbox and lens profile

BludzUK, 2022-05-08:
Thanks
BBL - https://www.dropbox.com/s/93i7p5ypbm0ej6i/btfl_001.bbl?dl=0
Video - https://www.dropbox.com/s/gg1c3utu0cinmwf/magicfc1.mp4?dl=0
Lens Profile - Happens whether you select one or not but I use Blackmagic 4k 7.5 Laowa 4096x2160

BludzUK, 2022-05-08:
Okay okay, maybe noob error but the settings for stabalistation had changed to fixed camera and zoomed in so much it looked llike a blank screen. hahahahahah
Seem to have it going now. Looking forward to messing with it.

While I have you... any news on your gyrologger? really don't want to use a naked gopro and fc and others seem a bit hit and miss for me so far.

AdrianEddy, 2022-05-09:
Ok, I'm glad it works now.

About the gyrologger ask DusKing1 on Discord

closed 2022-05-09 by AdrianEddy