gyroflow/gyroflow #339: Render queue enhancement/bug (maintain resolution)

p-masek, 2022-05-09:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Not sure if it is enhancement or bug but I have below request/problem:

1. I load first video file into Gyroflow, setup export settings - adjust codec, resolution, bitrate.
2. I add video file into render queue, everything is fine as expected
3. When I drag&drop other video files into the render queue - all export adjustments from first clip are taken over to the new clips EXCEPT resolution (codec, bitrate is per first clip; resolution is still on its default/original). To adjust resolution I need to edit&save each clip separately which can be very time consuming. 
ENHANCEMENT: newly added clips would take over all 3 export parameters from first clip - codec, resolution, bitrate.

AdrianEddy, 2022-05-09:
I don't think it should work this way, how can Gyroflow know that you want to set the same resolution to all files? I'm pretty sure if I do that, we'll get another bug report - `Gyroflow changes resolution randomly in the render queue`
Bitrate is also not based on first file, it is set to input file bitrate

p-masek, 2022-05-09:
Take a look on screenshots below:

First clip imported into GF, Export settings on default:

[image]

Export settings updated - codec, resolution, bitrate:

[image]

Clip added into queue:

[image]

Other clips added using drag&drop - as you can see - codec, bitrate is per first clip. Resolution is on default

[image]

Regarding possible bug `Gyroflow changes resolution randomly in the render queue` . Why "randomly"? It would be same resolution as per first clip which make perfect sense to me (if you do not want to go by way of using some sort of templates).

I just want use same settings on every clip in queue and it is almost perfect now except resolution.

Or is there any way how to applies same settings on all clips in render queue? My point is to speedup workflow as much as possible but now - in case of  lot of clips - you would need to edit export settings (but just resolution) of each file (it is necessary as x265 does not support 5,3K videos from GoPro).

p-masek, 2022-05-16:
So is there anyway to use same settings on every clip in render queue without indiviual changes of every clip?

p-masek, 2022-05-16:
I see new item #343 so closing this request.

closed 2022-05-16 by p-masek