gyroflow/gyroflow #349: Insta Go 2 / caddx peanut lens profile not applied

ps-igel, 2022-05-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.0 rc5

### What operating system are you using?

Windows 10

### What GPU are you using?

GTX 1060

### What happened?

Applying the lens profile Insta 360 GO 2 2688x2688 to footage of my Caddx Peanut used to work prior to Gyroflow 1.0 RC5. 

Since RC5 it does not apply the lens correction, leaves the fish-eye and just paints a black box around it. 

I also tried version 1.0.1 as well as the lastest automatic built.

Here is an example screenshot:

RC4 - working:
[image]

RC5 - not working:
[image]

Log output below from latest built from starting up to loading the profile (probably not very helpful).

### Relevant log output

```shell
15:26:11 [WARN] Binding on animation is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
15:26:11 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
15:26:11 [WARN] Binding on popup is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
15:26:11 [WARN] Binding on indicator is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
15:26:11 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
15:26:12 [INFO] Lens profiles directory: "\\\\?\\C:\\Users\\xxx\\Downloads\\gyroflow_latest\\Gyroflow-windows64\\camera_presets"
15:26:12 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("NVIDIA CUDA") ["NVIDIA GeForce GTX 1060 6GB"]
15:26:12 [INFO] OpenCL Platform: NVIDIA CUDA, Device: NVIDIA Corporation NVIDIA GeForce GTX 1060 6GB
15:26:12 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
15:26:12 [INFO] Loaded 3025 lens profiles in 177.117ms
15:26:12 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
[src\rendering\mod.rs:50] gpu_type = NVIDIA
15:26:25 [DEBUG] (18) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
15:26:25 [DEBUG] (19) MDKPlayer: [in m_player->onEvent] "thread.video" ""
15:26:25 [DEBUG] (20) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
15:26:25 [DEBUG] (20) MDKPlayer: [in m_player->onEvent] "decoder.audio" "FFmpeg"
15:26:25 [DEBUG] (20) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
15:26:25 [DEBUG] (21) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Direct3D11
15:26:25 [DEBUG] (21) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x14c93a3afd0) QSize(722, 722)
15:26:25 [DEBUG] (19) MDKPlayer: [in m_player->onEvent] "decoder.video" "AVCodec.D3D11"
15:26:25 [DEBUG] (21) MDKPlayer: [in m_player->onEvent] "render.video" "1st_frame"
15:26:25 [INFO] surface size: 1080x1080
15:26:25 [DEBUG] (21) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Direct3D11
15:26:25 [DEBUG] (21) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x14cbd011190) QSize(1080, 1080)
```

AdrianEddy, 2022-05-19:
Looks like you have lens correction strength set to 0? Set it to 100%

ps-igel, 2022-05-19:
Well yeah, that explains a lot. Did not notice the option because it was not there before. Thanks, issue closed then.

closed 2022-05-19 by ps-igel