gyroflow/gyroflow #351: Gyroflow crashes in Linux using hlg3 rec.2020

Bearsaerker, 2022-05-22:
So this is quite weird. I was puzzled why some files work, others don't. But i figured out, that on a zv-e10 every video recorded in slog2 is capable of being stabilized, while every single file recorded in hlg3 with rec.2020 will crash the application. I'm really a noob at reporting and I don't know which information would be demanded here. I use the newest version of gyroflow (1.0.1)

AdrianEddy, 2022-05-23:
Please upload the video file

Bearsaerker, 2022-05-23:
I uploaded the files, the xml and the terminal output I get when starting gyroflow from terminal. If something else is needed, I'll give my best to provide it! https://1drv.ms/u/s!ApRFYBePafhsgbUBcwwTeo5jBpUAlw?e=FLkVxg

AdrianEddy, 2022-05-23:
can you also provide `gyroflow.log` file after crash?

Bearsaerker, 2022-05-23:
that is the log file i get, when gyroflow crashes when importing hlg3 [gyroflow.log](https://github.com/gyroflow/gyroflow/files/8753913/gyroflow.log)

AdrianEddy, 2022-05-23:
@wang-bin can you take a look? I tested on Windows and macOS and it works so looks like issue is only on Linux.

wang-bin, 2022-05-25:
I use snprintf to generate shader code. snprintf is locale sensitive, an int value `1000` may becomes `1,000`, an extra comma is added, which is unexpected

wang-bin, 2022-05-26:
fixed in the latest build @AdrianEddy

AdrianEddy, 2022-05-31:
@Bearsaerker please test in latest dev build https://gyroflow.xyz/devbuild/?autodownload

Bearsaerker, 2022-05-31:
@AdrianEddy Just tested it, works now like a charm! Thank you!