gyroflow/gyroflow #353: Editable fields not working on Linux

bjin, 2022-05-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

master as well as 1.0.1(appimage)

### What operating system are you using?

Arch Linux

### What GPU are you using?

Intel UHD Graphics 620

### What happened?

I might be missing something essential, but it seems that on Arch Linux, all editable fields of Gyroflow UI is somehow not working.

[image]

Note that there is no edit button on the right side of `---`, I also tried to click places here and there, no luck. Without specifying Camera parameters I couldn't even save the calibrated JSON file (without a filename).

I tried both compiling from git master and the AppImage for 1.0.1 release.

The `qt6-base` version is `6.3.0` on Arch Linux.

### Relevant log output

```shell
MDK 0.14.2 (git 5d3d513) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang14.0.1 GCC4.2.1; libc++14000; 09:20:09 Apr 30 2022
15:10:14 [WARN] Binding on animation is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
15:10:14 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
15:10:14 [WARN] Binding on popup is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
15:10:14 [WARN] Binding on indicator is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
15:10:14 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
15:10:15 [INFO] Lens profiles directory: "/home/bjin/src/gyroflow/resources/camera_presets"
15:10:15 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Intel(R) OpenCL HD Graphics") ["Intel(R) UHD Graphics 620 [0x3ea0]"]
15:10:15 [INFO] OpenCL Platform: Intel(R) OpenCL HD Graphics, Device: Intel(R) Corporation Intel(R) UHD Graphics 620 [0x3ea0]
[src/rendering/mod.rs:50] gpu_type = Intel
15:10:15 [INFO] Loaded 3025 lens profiles in 56.360ms
```

AdrianEddy, 2022-05-26:
Icons are missing, some dependency related to svg is missing, but I don't have experience with Arch to debug it

bjin, 2022-05-26:
> some dependency related to svg is missing

Is this a known issue? Could this be the root cause?

> You should be able to export the profile even without any info, what's exactly the issue with that?

Well, I couldn't specify the filename (of the .json file) in the dialog window for the destination directory. So I somehow assumed that the filename is actually generated from camera parameters specified with editable fields above. Now, I realized that  could be wrong on this.

Anyway, I find a workaround to save the json file. Just create an empty json file in the destination directory, and select it to overwrite it.

bjin, 2022-05-26:
Yes, `strace` shows that `gyroflow` tries to read svg files under `/usr/share/pixmaps/`, so it's a packaging problem. The package maintainer should install those `.svg` files to `/usr/share/pixmaps/` in other distro.

However, the problem remains for developers, those svg file is missing as well for `cargo run`. Is it possible to search for those svg files dynamically, instead of reading from system resource folder?

Anyway, thanks for explaining this and point the right direction.

closed 2022-05-26 by bjin

AdrianEddy, 2022-05-26:
Gyroflow should use embedded .svg files, they are in the binary itself - not on the filesystem.

Icons are stored in .qrc, and then this qrc path is provided to `QIcon::setThemeSearchPaths` in `gyroflow.rs`

The issue to figure out here is why it doesn't use these embedded icons from .qrc but tries to load from filesystem

bjin, 2022-05-26:
I see. It make sense now.