gyroflow/gyroflow #354: Gyrodata Sony FDR-X1000v

Noctaro, 2022-05-27:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Hi all! Thank you for the amazing work. I wonder if it is possible to use gyrodata from the Sony FDR-X1000V to stabilize. 
I can import footage but it does not show me gyro traces. Anyways Gyroflow says it detects data in the video file. Let me know if you need a sample video file or any other data to check it.

[image]

AdrianEddy, 2022-05-27:
I don't think FDR-X1000V records gyro data.
You can send a sample file though, I'll verify

AdrianEddy, 2022-06-08:
I'm 99% sure it doesn't record gyro data, if you send me a sample file I'll verify, but there's very little chance for that.

closed 2022-06-08 by AdrianEddy