gyroflow/gyroflow #359: MXF files not loading into GyroFlow

BabaTom, 2022-05-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

RC 1.0.1

### What operating system are you using?

12.4

### What GPU are you using?

2.3 GHz 8-Core Intel Core i9

### What happened?

Just installed latest 30 May '22 dev build; when I drop .mxf (Sony FX6) files in to Gyroflow, however, I am getting a long spinning wheel, but that's it. File never opens in GyroFlow. Anything changed that is doing this? Should I be using RC 1.01 instead? Will greatly appreciate any guidance as I need to hustle with this material! Thank you.
Dev build is RC 1.0.1 (gh517) btw
- Am on 15 inch 2019 8 core MBP, 32gb RAM, running latest Monterrey
- JUST REALIZED gh519 is the very latest dev build; installed and tried with it too, but exactly same problem, unfortunately
- And now, I also have tried with the RC 1.0.1 'official release'; same problem - just a spinning wheel and files do not open.
FAST HELP OR A LINK TO VERSION YOU KNOW WILL WORK WITH MY MXF FILES WILL BE HUGELY APPRECIATED.
Thank you!

### Relevant log output

```shell
None
```

AdrianEddy, 2022-05-31:
does it work in rc5?

AdrianEddy, 2022-05-31:
You can download all previous versions from here: https://github.com/gyroflow/gyroflow/releases under `Assets`
I just need to know if it worked before, to pinpoint the issue

BabaTom, 2022-05-31:
> 1.0.5 that is posted is a March version; you guys fixed the audio issue in a dev build around 22 April and that worked AOK; I don’t see where I can grab that dev build anywhere as if I go all the way back to the March version, I lose all my audio, thereby making triple work for me (!). CAn you point me to a dev build that was after 22 April, please, pretty please, Adrian? Thank you. Tom

Files open AOK in 1.0.5, just loaded the March ‘official release and tested it. And as I wrote earlier, later dev builds that I used in April (before 1.0.1, that is) also worked AOK for me with Sony FX6 .mxf files.

Will be most grateful if you all may be able to solve this issue.

Thank you. Tom

AdrianEddy, 2022-05-31:
How big is this .mxf file? does it happen with all .mxf files?

BabaTom, 2022-05-31:
> I tried a wide variety, from well under a gig to three gigs; same result and all open aok in 1.0.5

Also just tried latest, today’s, 31 May build, to no avail. Thanks. Tom

AdrianEddy, 2022-05-31:
Please share a test file, I just tested on macbook m1 air with .mxf files from FX6 and they worked in gh521

BabaTom, 2022-06-01:
WeTransferred you two .MXF files just now. As I wrote, I am capturing with FX6 using exactly same parameters as I always do and these open and can be stabilized fine in 1.0.5; this included 1.0.5 development builds until just before 1.0.1 came out. Thanks, Adrian! Best, Tom

AdrianEddy, 2022-06-01:
Hmm, they open fine on Macbook M1 Air and Macbook M1 Pro with latest macOS Monterey.
Can you also upload `gyroflow.log` file?
You can also try this:
1. Right click on Gyroflow -> Open package contents -> Contents -> MacOS
2. Open terminal and type `export NO_OPENCL=1` and hit enter
3. Drag and drop `gyroflow` file into that terminal and hit enter
4. Try to load the file

BabaTom, 2022-06-01:
Managed to open GyroFlow using Terminal, as you instructed below, but still no luck getting it to open .mxf files. Log will probably tell you I am trying with latest dev build (gh 524).

Here’s log file you requested.

Let me know whatever else you need me to do from this end, please.

Thank you.

Tom

_______________________
Thomas Dallal
www.thomasdallal.com
***@***.***
+972 54 981 1854
+972 59 764 3931
+1 646 801 9640

BabaTom, 2022-06-01:
> @wang-bin <https://github.com/wang-bin> can you take a look? gyroflow.log <https://github.com/gyroflow/gyroflow/files/8815148/gyroflow.log>
> @BabaTom <https://github.com/BabaTom> also try this - same steps but instead of export NO_OPENCL=1 type export MDK_DECODERS=FFmpeg

Kindly re-type Terminal command - it does not work when I tyope or cut and paste it into Terminal

AdrianEddy, 2022-06-01:
@BabaTom can you also share log from the last working version?

BabaTom, 2022-06-01:
Just overwrote 1.0.1 latest build with this old (March) version of 1.0.5, did a quick test with an .mxf file that worked AOK, and here attached below now is the log file from that 1.0.5 installation.

wang-bin, 2022-06-01:
it's my bug in a recent change. try the latest build

BabaTom, 2022-06-01:
Thank you for youry reply, wang-bin, but latest build (gh527) still does not open my .mxf files. As Adrian requested, I am attaching here the logfile I just made using rc1.0.5, which opens these .mxf files without any issue. Also, Adrian, export MDK_DECODERS=FFmpeg in Terminal also did not work. Thank you, Tom
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/8816089/gyroflow.log)

AdrianEddy, 2022-06-01:
It's updated in build gh528, please check

BabaTom, 2022-06-01:
Wow, Adrian and all; IT WORKS! Just tested my first file, will be a long night of more but its a good start.

First file I tested - without changing any settings at all in the app - looks better than any result I’ve previously had with GyroFlow.

Now to test batch rendering ProRes files tonight.

Also, I saw someone else again raising the Timecode issue I mentioned some months ago; anything you can do to preserve, or ‘pass-thru’, time code, as you so deftly did with audio for these .mxf files a while back, will make GyroFlow fit for any possible professional use. If you find the bandwidth, that is!

What can I say, but THANK YOU and KEEP UP THE FABULOUS WORK. I will keep trying to feedback any useful input I can, of course.

Kind regards,

Tom
_______________________
Thomas Dallal
www.thomasdallal.com
***@***.***
+972 54 981 1854
+972 59 764 3931
+1 646 801 9640

AdrianEddy, 2022-06-01:
I'm glad it's fixed :)
Thanks @wang-bin!

closed 2022-06-01 by AdrianEddy