gyroflow/gyroflow #363: ERROR: Validation Error (0x8876086C)

doc123307, 2022-06-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.0.1 and dev build 2022-06-01T14:52:33Z (333MB)

### What operating system are you using?

Windows 10 v.21H2 (19044.1706)

### What GPU are you using?

intel core i7 2670qm 2.20ghz

### What happened?

As soon as I press the "export" button, the process stops and an error window pops up.
My laptop specifications:
intel core i7 2670qm 2.20ghz
NVIDIA GeForce GT 630M
-------------------------------------
I tried it on another xiaomi mi notebook pro 2017 laptop - nothing happened.
-------------------------------------
Works only on a desktop PC.
-------------------------------------
I attach screenshots:
https://drive.google.com/drive/folders/1Dwdy_5nb1vHZdq3rg8Rd7RHoEbWW4xpA?usp=sharing

### Relevant log output

```shell
16:59:11 [INFO] [MDK] MDK 0.14.2 (git 9bd409e) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Windows0x0600 x86_64; MSVC1931; MSVCRT14.31.31103.0/msvcp140.dll; 13:15:22 May  3 2022
16:59:11 [DEBUG] (1) mdk: [MDK] 3mdk verify key signature ok
16:59:11 [DEBUG] (1) mdk: [MDK] 3mdk license key restrictions: OpenSource,
16:59:11 [DEBUG] (1) mdk: [MDK] user module: D:\Program Files\Gyroflow-windows64\Gyroflow.exe
16:59:11 [DEBUG] (1) mdk: [MDK] user module: D:\Program Files\Gyroflow-windows64\Gyroflow.exe
16:59:11 [DEBUG] (1) mdk: [MDK] 3mdk license key for app: gyroflow
16:59:11 [WARN] Binding on animation is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
16:59:11 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
16:59:11 [WARN] Binding on popup is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
16:59:11 [WARN] Binding on indicator is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
16:59:11 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
16:59:14 [INFO] Lens profiles directory: "\\\\?\\D:\\Program Files\\Gyroflow-windows64\\camera_presets"
16:59:14 [INFO] Latest version: v1.0.1, current version: 1.0.1
16:59:14 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Intel(R) OpenCL") ["Intel(R) Core(TM) i7-2670QM CPU @ 2.20GHz"]
16:59:14 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("NVIDIA CUDA") ["GeForce GT 630M"]
16:59:14 [INFO] OpenCL Platform: NVIDIA CUDA, Device: NVIDIA Corporation GeForce GT 630M
16:59:14 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
16:59:14 [INFO] Loaded 3031 lens profiles in 213.295ms
16:59:14 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
16:59:14 [WARN] QWindowsWindow::setGeometry: Unable to set geometry 1371x706+0+31 (frame: 1387x745-8+0) on QQuickWindowQmlImpl_QML_173/"" on "\\.\DISPLAY1". Resulting geometry: 1370x706+0+31 (frame: 1386x745-8+0) margins: 8, 31, 8, 8)
16:59:17 [DEBUG] (1) mdk: [MDK] Registered audio backends: XAudio2 DSound
16:59:17 [DEBUG] (1) mdk: [MDK] XAudio2_9.dll
16:59:17 [DEBUG] (1) mdk: [MDK] Build with XAudio2 from Win8+ SDK
16:59:17 [DEBUG] (1) mdk: [MDK] Try symbol 'XAudio2Create' from Windows SDK dll
16:59:17 [DEBUG] (1) mdk: [MDK] Registered audio backends: XAudio2 DSound
16:59:17 [DEBUG] (1) mdk: [MDK] 000001627D11F9C0__cdecl mdk::abi::Player::Player(void)
16:59:17 [INFO] [MDK] 0000016208868430 MediaControl.prepare(0, ...) D:/Видео/GX010253.MP4
16:59:17 [DEBUG] (1) mdk: [MDK] 0000016276F85B10 FrameReaderImpl.start(0, ...)
16:59:17 [DEBUG] (10) mdk: [MDK] start frame reader thread: 5724
16:59:17 [DEBUG] (10) mdk: [MDK] try to load av module: D:\Program Files\Gyroflow-windows64\ffmpeg-5.dll
16:59:17 [DEBUG] (10) mdk: [MDK] try to load av module: D:\Program Files\Gyroflow-windows64\avformat-59.dll
16:59:17 [DEBUG] (10) mdk: [MDK] Selected avformat runtime version: 59.16.100 (build: 59.20.101), license: GPL version 2 or later
[... 750 lines cut ...]
16:59:46 [DEBUG] (1) mdk: [MDK] 000001627D11F9C0__cdecl mdk::abi::Player::~Player(void)
16:59:46 [INFO] [MDK] setNext from -1 url:
16:59:46 [INFO] [MDK] 0000016276F85B10 void __cdecl mdk::abi::FrameReaderImpl::stop(void), current state: 2
16:59:46 [DEBUG] (1) mdk: [MDK] 0000016276F85B10 bool __cdecl mdk::abi::FrameReaderImpl::unload(void), current status 0x124, invalid: 0, loading: 0, loaded: 4, buffering: 0, seeking: 0, prepared: 256, EOF: 0
16:59:46 [DEBUG] (1) mdk: [MDK] 00000162092D7960 request avformat input interrupt
16:59:46 [DEBUG] (1) mdk: [MDK] 00000162095C7E38 request to interrupt io operation
16:59:46 [DEBUG] (1) mdk: [MDK] __cdecl mdk::abi::MediaPlayer::Private::~Private(void)
16:59:46 [DEBUG] (1) mdk: [MDK] 0000016208868430 void __cdecl mdk::abi::MediaControlPush::setState(enum mdk::abi::State)@651 requested state 0=>0, current state 2
16:59:46 [DEBUG] (12) mdk: [MDK] EOS audio frame of track 0
16:59:46 [DEBUG] (10) mdk: [MDK] free custome avio: 00000162092E9680
16:59:46 [DEBUG] (10) mdk: [MDK] 00000162095C7E20 close url: D:/Видео/GX010253.MP4
16:59:46 [DEBUG] (12) mdk: [MDK] audio stream#1 decoding loop is finished
16:59:46 [DEBUG] (10) mdk: [MDK] stop by user. clear packet queues and send eof packets for each thread...
16:59:46 [DEBUG] (10) mdk: [MDK] FrameReader@0000016276F85B10 sending Packet::End to all packet queues. PacketIO@00000162095C75A0 ->url: D:/Видео/GX010253.MP4
16:59:46 [DEBUG] (10) mdk: [MDK] clear packet queue for stream 0
16:59:46 [DEBUG] (12) mdk: [MDK] __cdecl ThreadLocal<struct mdk::abi::ComInit>::Data::~Data(void) thread: 7956
16:59:46 [DEBUG] (11) mdk: [MDK] EOS video frame of track 0
16:59:46 [DEBUG] (10) mdk: [MDK] clear packet queue for stream 1
16:59:46 [DEBUG] (11) mdk: [MDK] video stream#0 decoding loop is finished
16:59:46 [DEBUG] (10) mdk: [MDK] 0000016276F85B10 updateState: 0
16:59:46 [DEBUG] (1) mdk: [MDK] 0000016208867EA0 void __cdecl mdk::abi::MediaControlPush::setState(enum mdk::abi::State)@651 requested state 0=>0, current state 0
16:59:46 [DEBUG] (1) mdk: [MDK] player is destroyed
16:59:46 [DEBUG] (1) mdk: [MDK] __cdecl mdk::abi::VideoRenderer::Private::~Private(void) @114
16:59:46 [DEBUG] (1) mdk: [MDK] __cdecl mdk::abi::VideoRenderer::Private::~Private(void) @114
```

doc123307, 2022-06-01:
> Do you have latest GPU drivers installed? GeForce GT 630M is a very old
> GPU and it most likely doesn't support GPU encoding.
> You can try x264, but if that doesn't help, uncheck GPU encoding and it
> should work using your CPU

As soon as I removed the "hardware acceleration" - everything worked.
Strange, on a xiaomi laptop 4 GB. video memory and processor is good, but
for some reason it does not work with a checkmark.

AdrianEddy, 2022-06-01:
Can you send me the log from latest dev build? https://gyroflow.xyz/devbuild/?autodownload

doc123307, 2022-06-02:
Of course.
Send file and screenshot

doc123307, 2022-06-02:
11:41:04 [INFO] [MDK] MDK 0.14.2 (git 8c233cf) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Windows0x0600 x86_64; MSVC1932; MSVCRT14.32.31326.0/msvcp140.dll; 15:28:48 Jun  1 2022
11:41:04 [DEBUG] (1) mdk: [MDK] 3mdk verify key signature ok
11:41:04 [DEBUG] (1) mdk: [MDK] 3mdk license key restrictions: OpenSource,
11:41:04 [DEBUG] (1) mdk: [MDK] user module: C:\Users\AVasilych\Documents\Gyroflow-windows64\Gyroflow.exe
11:41:04 [DEBUG] (1) mdk: [MDK] user module: C:\Users\AVasilych\Documents\Gyroflow-windows64\Gyroflow.exe
11:41:04 [DEBUG] (1) mdk: [MDK] 3mdk license key for app: gyroflow
11:41:07 [WARN] Binding on animation is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
11:41:07 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
11:41:07 [WARN] Binding on indicator is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
11:41:07 [WARN] Binding on popup is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
11:41:07 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
11:41:07 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
11:41:10 [INFO] Lens profiles directory: "\\\\?\\C:\\Users\\AVasilych\\Documents\\Gyroflow-windows64\\camera_presets"
11:41:10 [INFO] Loaded 3030 lens profiles in 122.773ms
11:41:11 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Intel(R) OpenCL") ["Intel(R) UHD Graphics 620", "Intel(R) Core(TM) i7-8550U CPU @ 1.80GHz"]
11:41:11 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("NVIDIA CUDA") ["GeForce MX150"]
11:41:11 [INFO] OpenCL Platform: NVIDIA CUDA, Device: NVIDIA Corporation GeForce MX150
11:41:11 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
11:41:12 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
11:41:21 [DEBUG] (1) mdk: [MDK] Registered audio backends: XAudio2 DSound
11:41:21 [DEBUG] (1) mdk: [MDK] XAudio2_9.dll
11:41:21 [DEBUG] (1) mdk: [MDK] Build with XAudio2 from Win8+ SDK
11:41:21 [DEBUG] (1) mdk: [MDK] Try symbol 'XAudio2Create' from Windows SDK dll
11:41:21 [DEBUG] (1) mdk: [MDK] Registered audio backends: XAudio2 DSound
11:41:21 [DEBUG] (1) mdk: [MDK] 000001EA76FD4CE0__cdecl mdk::abi::Player::Player(void)
11:41:21 [INFO] [MDK] 000001EA21619580 MediaControl.prepare(0, ...) D:/GOPRO_TEST/GX010267.MP4
11:41:21 [DEBUG] (1) mdk: [MDK] 000001EA73B87F70 FrameReaderImpl.start(0, ...)
11:41:21 [DEBUG] (10) mdk: [MDK] start frame reader thread: 8900
11:41:21 [DEBUG] (10) mdk: [MDK] try to load av module: C:\Users\AVasilych\Documents\Gyroflow-windows64\ffmpeg-5.dll
11:41:21 [DEBUG] (10) mdk: [MDK] try to load av module: C:\Users\AVasilych\Documents\Gyroflow-windows64\avformat-59.dll
11:41:21 [DEBUG] (10) mdk: [MDK] Selected avformat runtime version: 59.16.100 (build: 59.24.100), license: GPL version 2 or later
11:41:21 [DEBUG] (10) mdk: [MDK] Selected avformat runtime configuration: --extra-version=avbuild --disable-doc --disable-debug --enable-shared --enable-runtime-cpudetect --enable-libmfx --disable-ptx-compression --enable-mediafoundation --disable-postproc --toolchain=msvc --windres=llvm-windres --extra-windresflags='--target=x86_64-pc-windows-msvc -I/c/Program\ Files/Microsoft\ Visual\ Studio/2022/Enterprise/VC/Tools/MSVC/14.31.31103/ATLMFC/include -I/c/Program\ Files/Micros
11:41:21 [DEBUG] (10) mdk: [MDK] oft\ Visual\ Studio/2022/Enterprise/VC/Tools/MSVC/14.31.31103/include -I/c/Program\ Files\ \(x86\)/Windows\ Kits/NETFXSDK/4.8/include/um -I/c/Program\ Files\ \(x86\)/Windows\ Kits/10/include/10.0.22000.0/ucrt -I/c/Program\ Files\ \(x86\)/Windows\ Kits/10/include/10.0.22000.0/shared -I/c/Program\ Files\ \(x86\)/Windows\ Kits/10/include/10.0.22000.0/um -I/c/Program\ Files\ \(x86\)/Windows\ Kits/10/include/10.0.22000.0/winrt -I/c/Program\ Files\
11:41:21 [DEBUG] (10) mdk: [MDK] \(x86\)/Windows\ Kits/10/include/10.0.22000.0/cppwinrt' --enable-pic --extra-cflags='-Zi -FS -MD -guard:cf -I/d/a/avbuild/avbuild/tools/dep/include -D_WIN32_WINNT=0x0600' --extra-ldflags='-DEBUG -guard:cf -OPT:REF -SUBSYSTEM:CONSOLE -NODEFAULTLIB:libcmt' --enable-small --disable-filters --enable-filter='*null*,*fade,*fifo,*format,*resample,aeval,all*,atempo,pan,color*,convolution,crop,draw*,eq*,framerate,*_cuda,*_qsv,*_vaapi,*v4l2*,hw*,scale,v
11:41:21 [DEBUG] (10) mdk: [MDK] olume' --disable-muxers --disable-encoders --disable-decoders --enable-decoder='*sub*,movtext,*web*,aac*,*ac3*,alac*,ape,ass,av1*,cc_dec,cook,dca,dnxhd,eac3*,exr,truehd,ff*,*yuv*,flv,flac,gif,h26[3-4]*,hevc*,hap,mp[1-3]*,prores,*peg*,mlp,mpl2,nellymoser,opus,pcm*,*png*,rawvideo,rv*,sami,srt,ssa,v210*,vc1*,vorbis,vp[6-9]*,wm*,wrapped_avframe' --disable-demuxers --enable-demuxer='*sub*,*ac3,*ac,*peg*,*web*,ape,ass,avi,concat,dnxhd,dts*,*dash*,*f
11:41:21 [DEBUG] (10) mdk: [MDK] lv,gif,hls,h264,hevc,kux,xv,matroska,mlv,mov,mp3,mxf,nsv,nut,ogg,pcm*,rawvideo,rt*p,spdif,srt,vc1,v210*,wav,*pipe,image2' --disable-parsers --enable-parser='*sub*,aac*,ac3,cook,dnxhd,flac,h26[3-4],hevc,m*,opus,rv*,vc1,vorbis,vp [... cut ...]
video stream#0 decoding loop is finished
11:41:49 [DEBUG] (10) mdk: [MDK] 000001EA73B87F70 updateState: 0
11:41:49 [DEBUG] (1) mdk: [MDK] 000001EA21615DE0 void __cdecl mdk::abi::MediaControlPush::setState(enum mdk::abi::State)@647 requested state 0=>0, current state 0
11:41:49 [DEBUG] (1) mdk: [MDK] player is destroyed
11:41:49 [DEBUG] (1) mdk: [MDK] __cdecl mdk::abi::VideoRenderer::Private::~Private(void) @114
11:41:49 [DEBUG] (1) mdk: [MDK] __cdecl mdk::abi::VideoRenderer::Private::~Private(void) @114
[image]

AdrianEddy, 2022-06-02:
Does GPU encoding to x264 work?

doc123307, 2022-06-02:
No, it does not work. Only there is a tick "hardware acceleration"

AdrianEddy, 2022-06-02:
Ok, `GeForce GT 630M` is over 10 years old so it doesn't support any GPU acceleration, you have to stick with CPU rendering

doc123307, 2022-06-02:
And what about Xiaomi Mi Notebook Pro?
Intel Core i7-8550U, L3 8 MB
NVIDIA GeForce® MX150 2 GB GDDR5

AdrianEddy, 2022-06-02:
GeForce MX150 also doesn't support any hardware acceleration:
it uses GP108 chipset which doesn't have any encoders: https://en.wikipedia.org/wiki/Nvidia_NVENC
[image]

closed 2022-06-02 by doc123307