gyroflow/gyroflow #368: lens profile reverts to GoPro

rpossum, 2022-06-03:
using render queue... loading mp4 files, with my Samyang 12mm lens profile... and a GoPro file for gyro... I kept getting a huge lens distortion on outputs, even though it looked good in GyroFlow preview. Then I realized, it keeps reverting automatically to a GoPro Hero 10 Black lens profile... if I choose one from the queue and EDIT, it reverts to the GoPro lens profile every time... I fix it, then it reverts. The only way top get good output is to set lens correction strength to 0

AdrianEddy, 2022-06-06:
It should be fixed in latest dev build (gh538), please test and let me know https://gyroflow.xyz/devbuild/?autodownload

closed 2022-06-07 by AdrianEddy

rpossum, 2022-06-10:
It's fixed! Thank you!
But it does seem like there's a new issue... the sync point dots in this build are not aligning to anything... they are just a grid
[image]
It's stabilizing well, but those dots don't tell you much like they used to

AdrianEddy, 2022-06-10:
This is expected, it's called Dense Optical Flow, it's just a different kind of optical flow.

rpossum, 2022-10-11:
ok awesome
thanks for the quick fix!