gyroflow/gyroflow #371: opencl.dll missing

bigjoncoop, 2022-06-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.0.1

### What operating system are you using?

Windows 10 Virtual Machine 

### What GPU are you using?

AMD 5450

### What happened?

Since my last issue is not resolved with gyroflow will not being able to load a video my only other option is to run it in a virtual machine.

When I try to run gyro flow I get an error that says opencl.dll cannot be found. I've scoured Google trying to find a fix for opencl.dll in a virtual machine and haven't been able to find anything useful.

I tried downloading the dll and putting it into the gyro for folder but I get the same error

AdrianEddy, 2022-06-05:
opencl.dll is included in graphic card drivers, it's not an issue in Gyroflow. You can also run Gyroflow with `NO_OPENCL=1` environment variable set

closed 2022-06-05 by AdrianEddy

nu11ptr, 2022-11-12:
FYI, I still get this error with `NO_OPENCL=1` set in my environment

AdrianEddy, 2022-11-15:
@nu11ptr It is *strongly* recommended to install newest drivers for your GPU, but if that doesn't work for you, you can try this one 
[OpenCL.zip](https://github.com/gyroflow/gyroflow/files/10009182/OpenCL.zip)