gyroflow/gyroflow #375: Error with ffmpeg when exporting

davenewt, 2022-06-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.0.1

### What operating system are you using?

MacOS 12.0.1

### What GPU are you using?

Radeon Pro 575 4 GB

### What happened?

When hitting export, the _stabilized file is created, but after 30 seconds or so, no frames have been processed and this error is shown:

An error occured: ffmpeg error: ffmpeg::Error(12912: Unknown error: 12912)
[hevc videotoolbox @ 0x7f8ba4a28c80] Error encoding frame: -12912
[aac @ 0x7f8ba4b5fd00] Qavg: nan
[heve_videotoolbox @ 0x7f8bb5aa6780] Error encoding frame: -12912
[aac @ 0x7f8ba4b28280] Qavg: nan
[hevc videotoolbox @ 0x7f8ba4a40600] Error encoding frame: -12912
[aac @ 0x7f8bb59b1d80] Qavg: nan
Theve videotoolbox @ 0x7f8ba4aaeac0] Error encoding frame: -12912
[aac @ 0x7f8ba4aac5c0] Qavg: nan
Theve videotoolbox @ 0x7f8c05541a401 Error encoding frame: -12912
[aac @ 0x7f8bb5aa7e80] Qavg: nan
[aac @ 0x7f8bb5aa8c00] Qavg: nan

[image]

davenewt, 2022-06-13:
PS this is on an Intel iMac, NOT m1. Thanks for taking a look, and for any advice! 🙂

closed 2022-06-13 by AdrianEddy

davenewt, 2022-06-13:
Ah, so you're telling me I neeeeeeeed a Mac Studio? 😉 Hehehe. OK thanks. Will run on m1 macbook or disable GPU encoding as you say! Thanks for the super quick response, and sorry for not realising that myself!

AdrianEddy, 2022-06-13:
You don't need Mac Studio, you can just as well use x264 with higher bitrate and it will be same quality :)