gyroflow/gyroflow #380: The app is crashing on startup

macrokernel, 2022-06-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.1.0

### What operating system are you using?

Linux Mint 20.2 Cinnamon

### What GPU are you using?

AMD Ryzen 5 4600H with Radeon Graphics (integrated graphics)

### What happened?

The app is crashing on every startup.

### Relevant log output

```shell
RUST_BACKTRACE=1 ./Gyroflow-linux64.AppImage
MDK 0.14.2 (git e2a85c2) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang14.0.5 GCC4.2.1; libc++14000; 00:00:12 Jun 13 2022

Gtk-Message: 17:02:46.335: Failed to load module "xapp-gtk3-module"

(gyroflow:1925291): dbind-WARNING **: 17:02:46.372: Couldn't register with accessibility bus: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
14:02:46 [WARN] Binding on animation is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
14:02:46 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
14:02:46 [WARN] Binding on popup is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
14:02:46 [WARN] Binding on indicator is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
14:02:46 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
14:02:46 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
14:02:47 [DEBUG] (2) gyroflow::util: Minidump uploaded: OK
14:02:48 [INFO] Lens profiles directory: "/tmp/.mount_GyroflvmKAdG/camera_presets"
14:02:48 [INFO] Loaded 3333 lens profiles in 235.617ms
14:02:48 [INFO] Latest version: v1.1.0, current version: 1.1.0 (gh570)
thread '<unnamed>' panicked at 'Platform::list: Error retrieving platform list: Unable to get platform id list after 10 seconds of waiting.', /root/.cargo/registry/src/github.com-1ecc6299db9ec823/ocl-0.19.3/src/standard/platform.rs:50:14
stack backtrace:
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
Rayon: detected unexpected panic; aborting
Crashdump written to /home/serge/bin/9520db6d-efea-4101-7fb3b987-6b60d75a.dmp
```

macrokernel, 2022-06-16:
Crash dump:
[9520db6d-efea-4101-7fb3b987-6b60d75a.dmp.gz](https://github.com/gyroflow/gyroflow/files/8919528/9520db6d-efea-4101-7fb3b987-6b60d75a.dmp.gz)

macrokernel, 2022-06-16:
Probably a duplicate of https://github.com/gyroflow/gyroflow/issues/154, but the fix mentioned in the issue does not help.

AdrianEddy, 2022-06-16:
Install opencl package for your system or add `export NO_OPENCL=1`

macrokernel, 2022-06-16:
Thanks a lot Adrian! Installing **mesa-opencl-icd** did help.
```shell
sudo apt install mesa-opencl-icd
```