gyroflow/gyroflow #381: GPLv3 License unfit

Johnnynator, 2022-06-16:
It is impossible right now to redistribute gyroflow in a compiled form and comply with the GPLv3 License.

The GPLv3 expects that the source code of every component (except “System Libraries” and “Standard Interfaces”) can be provided, which is not possible for mdk-sdk, since it is closed source.

I would recommend to relicense gyroflow under something like the [Apache License 2.0](https://spdx.org/licenses/Apache-2.0.html) (or some other copy left license that doesn't focus on the users' freedom) or replace mdk-sdk with some open source library.

AdrianEddy, 2022-06-21:
I'm hoping to replace mdk with Qt Multimedia once ffmpeg backend is implemented in Qt 6.4. 
We need GPLv3 to be able to use x264 and x265 encoders, so it should be good when I replace mdk with QtMultimedia
Do you know any other open-source libraries like mdk?

Johnnynator, 2022-06-21:
I think (lib)mpv can be leveraged to some degree to replace mdk, same with libvlc (But I have no personal experience using either 3 of them).

QtMultimedia might be a good bet, the ffmpeg backend is already in the current 6.4.0 alpha.

> We need GPLv3 to be able to use x264 and x265 encoders.

The used ffmpeg is GPLv2-or-later licensed and same is true for the x264 and x265 encoders, but this doesn't help, since GPLv2 has similar restrictions, just worded vastly different.

AdrianEddy, 2022-06-22:
I was looking into libmpv earlier but looks like it only supports OpenGL. I somehow assumed libvlc is basically the same, but looks like it's not the case and libvlc is more advanced with support to other graphic APIs too.
Looks like libvlc might work, I'll look into it, thanks

User-3090, 2022-07-23:
> I was looking into libmpv earlier but looks like it only supports OpenGL. I somehow assumed libvlc is basically the same, but looks like it's not the case and libvlc is more advanced with support to other graphic APIs too. Looks like libvlc might work, I'll look into it, thanks

Are you sure? mpv/mpv.net supports vulkan and d3d11 just fine and honestly VLC is IMHO super outdated and low quality. Also you would have access to high end scalers like FSRCNNX.

AdrianEddy, 2022-08-02:
@Johnnynator help me understand something, let's say I remove x264 and x265 and license Gyroflow as Apache 2.0, so the distribution won't contain any GPL code. But then, user is still able to replace the ffmpeg dlls, with the ones containing x264 and x265 on his own, correct? Is he required to build these dlls on his own from source? Can he use one of prebuilt ones, like from https://www.gyan.dev/ffmpeg/builds/ ? Or is every site/place providing prebuilt ffmpeg with x264 and x265 violating the GPL?

Johnnynator, 2022-08-03:
> @Johnnynator help me understand something, let's say I remove x264 and x265 and license Gyroflow as Apache 2.0, so the distribution won't contain any GPL code. But then, user is still able to replace the ffmpeg dlls, with the ones containing x264 and x265 on his own, correct? 

This would also be my understanding.

> I s he required to build these dlls on his own from source? Can he use one of prebuilt ones, like from https://www.gyan.dev/ffmpeg/builds/ ? Or is every site/place providing prebuilt ffmpeg with x264 and x265 violating the GPL?

FFMPEG builds licensed under GPLv3, as the gyan build are, can contain x264 and x265 and are perfectly fine to redistribute standalone, as done there.

>  let's say I remove x264

openh264 exists, but not sure how good it is compared to x264, don't think there is an alternative to x265 in ffmpeg.

Just curious, did you give QtMultimedia a look? The 6.4 betas do already exist.

AdrianEddy, 2022-08-03:
> and are perfectly fine to redistribute standalone, as done there.
So I cannot bundle ffmpeg GPL binary with other program, but I can provide standalone ffmpeg as binary for users to download themselves, right?
I'm essentially trying to figure out what would be the most convenient process for users while still being compatible with the GPL license.
Can I host the prebuilt GPLv3 ffmpeg on gyroflow server? Can Gyroflow itself download that ZIP with binary ffmpeg distribution for user? Does the user need to click "Agree" to the license before downloading? I'm guessing it can't download and extract the binaries in the background without user intervention? Where is the line?

Johnnynator, 2022-08-04:
I'm not sure where one would draw the line between "reasonable circumvention of some licensing rules" and license infirigement.

AdrianEddy, 2022-08-04:
Where did I say I want to circumvent anything? I'm trying to understand the rules exactly for the purpose of being compliant

Randrianasulu, 2022-12-02:
I think gpl3 is quite strict.

"c) You must license the entire work, as a whole, under this License to anyone who comes into possession of a copy. This License will therefore apply, along with any applicable section 7 additional terms, to the whole of the work, and all its parts, regardless of how they are packaged. This License gives no permission to license the work in any other way, but it does not invalidate such permission if you have separately received it."

https://www.gnu.org/licenses/gpl-3.0.en.html

For what exactly mdk-sdk used, as wrapper around ffmpeg or output proocessing?

For output processing libplacebo exist
https://github.com/haasn/libplacebo

Andrea-Paz, 2022-12-02:
Yes, I also tried to intervene in the bug in question. Better if the
author, as he says, brings everything to QT multimedia.

AdrianEddy, 2022-12-02:
mdk-sdk is used for zero-copy playback in GUI. I've looked into other libs but nothing else implements all 4 major graphic APIs (OpenGL, DirectX, Metal, Vulkan). The best bet currently is QtMultimedia in Qt 6.4 but it's not a trivial change
I'm also looking into implementing my own player utilizing libplacebo, but I need to do more research on that, specifically that libplacebo doesn't support Metal natively (only via MoltenVK) and I need to draw the video on MTLTexture in Metal context, I'm not sure if that's possible

Multimedia stuff is extremely complex and when you add 4 graphic APIs + processing with OpenCL or CUDA and interop between all this + different video formats it's all a nightmare to do properly.
It's not a simple OpenCV VideoCapture + imshow, Gyroflow is fast because a lot of care is taken to do everything on a low level in the GPU, but this makes it very complex.

I'm definitely pursuing ways to resolve this issue, but I didn't find ideal solution yet.

Randrianasulu, 2022-12-02:
@AdrianEddy thanks for explaining. Yes, making multimedia fast is not simple! Andrea and I  hack a bit on Cinelerra-GG (fork of cinelerra-HV), and I got she running on FreeBSD/NetBSD/Termux (!) so we wonder how far our compilation can proceed on Macos? But we lack any mac machine. If you have mac environment nearby it will be interesting to hear where our compilation stops (cinelerra-gg uses X11 for output, so this mean a lot of brew software like autotools and Xquartz, if it still exist, but developers tend to have more software/packages installed due to nature of their work, so may it will be not very huge effort on your part)

code is at
https://git.cinelerra-gg.org/git/?p=goodguy/cinelerra.git;a=summary

due to our history CinGG does have floating-point (32 bit float per channel) processing, not opencl/gl accelerated atm, but preview use opengl 2.1 era shaders over 8bpc textures/pbuffers (you can hack 32bpc textures easily if your OpenGL device support it. I did it one, with patch in our bug tracking system, and I hardly much of developer. Yes, Apple/Win push away from OpenGL but it still work for us, and we not really into non-x11 world. While surprizingly one user reported better performance via Xwayland on arm single board computer!)

A bit of offtopic, but Andrea found your software while looking at other image stabilizers (cingg contain software stabilizers too, not vidstab/ffmpeg) so I thought adding word about our existence will not hurt!

latenitefilms, 2022-12-13:
FWIW - I'm not sure if it's possible due to dependancies, but I'd personally love to see this project go to an MIT license.

latenitefilms, 2022-12-20:
FYI: This is how another proprietary app handles ffmpeg:

https://www.cdfinder.de/guide/22/22.7/neofinder_ffmpeg.html

kimariin, 2026-05-04:
As #1151 was closed in favour of this issue I will reiterate my primary concern here.

While it's great to hear that a replacement for mdk-sdk is [being worked on](https://github.com/AdrianEddy/gpu-video), the fact remains that today, Gyroflow binaries include an unsigned, unaudited closed-source blob whose presence is not signalled to users.

As Gyroflow is advertised - perhaps not on the website, but on the GitHub project page, on the [docs main page](https://docs.gyroflow.xyz/app), and of course through its LICENSE file - as free and open-source software, users would normally have no reason to suspect the presence of closed-source binaries in the release packages.

Could some kind of warning be added to the [release notes](https://github.com/gyroflow/gyroflow/releases/) and [download page](https://gyroflow.xyz/download)?

AdrianEddy, 2026-05-04:
It is upfront, it's stated in the README in the appropriate section about code, and it was there from the very first commit