gyroflow/gyroflow #383: crashing on Fedora Linux 36

sebnil92, 2022-06-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.1.0

### What operating system are you using?

Fedora 36

### What GPU are you using?

Nvidia GTX 1070 Ti

### What happened?

Nether 1.1.0 .tar.gz nor AppImage works on this version. I'm running 1.0.0 .tar.gz somewhat OK on this system. 1.0.0 AppImage does not work ether.

[dump from .tar.gz 835d1768-8fb1-4aad-0d85e08c-803cb24d.zip](https://github.com/gyroflow/gyroflow/files/8943509/835d1768-8fb1-4aad-0d85e08c-803cb24d.zip)
[dump from AppImage 6563149d-23ec-483b-e5626b97-4d21cbbd.zip](https://github.com/gyroflow/gyroflow/files/8943507/6563149d-23ec-483b-e5626b97-4d21cbbd.zip)

 

### Relevant log output

```shell
[sebnil@rex Gyroflow 1.1.0]$ ./gyroflow 
MDK 0.14.2 (git e2a85c2) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang14.0.5 GCC4.2.1; libc++14000; 00:00:12 Jun 13 2022
20:29:59 [WARN] Binding on animation is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
20:29:59 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
20:29:59 [WARN] Binding on popup is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
20:29:59 [WARN] Binding on indicator is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
20:29:59 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
20:29:59 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
20:30:00 [INFO] Lens profiles directory: "/home/sebnil/Downloads/Gyroflow 1.1.0/camera_presets"
20:30:00 [INFO] Loaded 3333 lens profiles in 18.246ms
20:30:00 [DEBUG] (7) gyroflow::util: Minidump uploaded: OK
thread '<unnamed>' panicked at 'Platform::list: Error retrieving platform list: Unable to get platform id list after 10 seconds of waiting.', /root/.cargo/registry/src/github.com-1ecc6299db9ec823/ocl-0.19.3/src/standard/platform.rs:50:14
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Rayon: detected unexpected panic; aborting
thread 'main' panicked at 'Platform::list: Error retrieving platform list: Unable to get platform id list after 10 seconds of waiting.', /root/.cargo/registry/src/github.com-1ecc6299db9ec823/ocl-0.19.3/src/standard/platform.rs:50:14
20:30:10 [ERROR] Failed to initialize OpenCL Platform::list: Error retrieving platform list: Unable to get platform id list after 10 seconds of waiting.
Crashdump written to /home/sebnil/Downloads/Gyroflow 1.1.0/835d1768-8fb1-4aad-0d85e08c-803cb24d.dmp
Aborted (core dumped)

[sebnil@rex Downloads]$ ./Gyroflow-linux64.AppImage 
MDK 0.14.2 (git e2a85c2) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang14.0.5 GCC4.2.1; libc++14000; 00:00:12 Jun 13 2022
Crashdump written to /home/sebnil/Downloads/6563149d-23ec-483b-e5626b97-4d21cbbd.dmp
```

AdrianEddy, 2022-06-20:
Install OpenCL package for your system (`pocl`)- See #380 or #137
OR use latest dev build, but you really should install OpenCL anyway https://gyroflow.xyz/devbuild/?autodownload

closed 2022-06-20 by AdrianEddy

sebnil92, 2022-06-22:
The `pocl` package did make gyroflow 1.0.1 start the render way faster. Before it was stuck on 0% for about 1 minute before starting rendering frames. Did not made any difference on 1.1.0 though.

After some more troubleshooting i found that the cuda driver `xorg-x11-drv-nvidia-470xx-cuda` from RPM fusion was needed to run 1.0.1 .tar.gz version. After i installed that it's now running almost flawless! Can this be added to the documentation? I could write it down and make a pull request if you want.

It does sometimes crash when i add another clip after exporting. I can post the logs for that when it happens next time!

AdrianEddy, 2022-06-22:
I'm glad you made it work, you can make a PR with the doc changes, thanks!