gyroflow/gyroflow #384: Dynamic zooming zooms in too much for lens correction strengths other than 0 and 100%

avsaase, 2022-06-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

72b8eb6a64d80292d1006277c4d4212865fca072

### What operating system are you using?

Windows 10

### What happened?

When selecting lens correction strengths other than 0 and 100% dynamic and fixed zooming zooms in too much. Take for example this clip without motion from the Firefly X Lite II on 2.7K 4:3. 

0% lens correction:
[image]

100% lens correction:
[image]

50% lens correction:
[image]

As you can see in the 50% lens correction example there is FOV "left on the table". This happens with all settings between 0 and 100% but seems to be maximized at or around 50%.

I increased the FOV to 1.5 to make it more obvious but it's not a bug in the `Safe area guide` because it also happens with FOV=1. It's also not related to the output size setting because I tried setting that to the input resolution but that didn't make a difference. I tried a couple of different lens profiles for this camera and setting, all with the same result. I recall seeing this with the Mobius Maxi 4K as well.

Here's the clip and `.gyroflow` file: https://mega.nz/folder/00NwGYwL#TAKzsDrmqTZ-FLzTkrvhIg

AdrianEddy, 2022-06-22:
I know, this is because I don't know how to calculate camera params when adding lens distortion back, so currently it's something approximated, see: https://github.com/gyroflow/gyroflow/blob/master/src/core/gpu/opencl_undistort.cl#L213

This will likely stay this way until someone smarter than me figures it out

AdrianEddy, 2023-09-22:
/bounty $150

aazam-gh, 2023-10-01:
/attempt #384

Rutik7066, 2023-12-25:
/attempt #384
got it.