gyroflow/gyroflow #386: GyroFlow is not Opening

eozuak, 2022-06-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

Gyroflow v1.1

### What operating system are you using?

Windows 10

### What GPU are you using?

Nvidea 1050 Ti

### What happened?

I downloaded the latest version of gyroflow. I extracted the files to the desktop. When I try to open the Gyroflow it suddenly shuts down.

AdrianEddy, 2022-06-26:
Is there any error message? Do you have latest windows updates and latest GPU drivers installed? Is there gyroflow.log file created? can you attach it?

AdrianEddy, 2022-07-08:
@eozuak any update?

jacobchow, 2022-07-11:
I also encountered the same problem.    Do you have latest windows updates and latest GPU drivers installed? ---- yes, I install latest GPU drivers for Intel and NVIDIA and DirextX, but Issue is unchanged.    gyroflow.log and crashdmp file are created, attach here
[logForIssue386.zip](https://github.com/gyroflow/gyroflow/files/9086072/logForIssue386.zip)

Looking forward to your reply and thank you for your contribution.

jacobchow, 2022-07-11:
> @eozuak any update?

Please note the previous comment.thanks

AdrianEddy, 2022-07-11:
> I also encountered the same problem. Do you have latest windows updates and latest GPU drivers installed? ---- yes, I install latest GPU drivers for Intel and NVIDIA and DirextX, but Issue is [...]

Edit `Gyroflow_with_console.bat` and add `set NO_OPENCL=1` as second line. Then run `Gyroflow_with_console.bat`

AdrianEddy, 2022-09-30:
Could you please test latest dev build? Qt was updated to new version so I'm curious if that fixed it https://gyroflow.xyz/devbuild/?autodownload