gyroflow/gyroflow #390: Blackmagic pocket cinema camera gyro support

OlegStelmakh, 2022-06-30:
Since blackmagic design released v7.9 firmware for all their bmpcc cameras (4k/6k/6k pro/6kg2) with gyrodata written into files, could gyroflow get a support to stabilize bmpcc footages?

AdrianEddy, 2022-06-30:
yes, this is being worked on

adinbied, 2022-07-03:
I'm sure some of this has already been figured out but figured I'd share what I'd discovered. I've found the data in each BRAW file in a hex editor and narrowed down the gyroscope and acceleration numbers but it's in some unknown hex encoding scheme [image]
I've figured out the 'mogy' (14 6D 6F 67 79)data is from the gyroscope and the 'moac' is from the accelerometer.
For the gyroscope data, theres a couple of interesting patterns that lead me to think its 3 sets of 3 byte values, seperated by 1 byte of header/padding - although could be off base.

AdrianEddy, 2022-07-11:
It's done in https://github.com/AdrianEddy/telemetry-parser/commit/f79d089038ce25f1f22af77ec56431ddf24621de
Available already in latest dev build: https://gyroflow.xyz/devbuild/?autodownload

closed 2022-07-11 by AdrianEddy

shaheedmalik, 2022-08-07:
So will it load Braw files? I tried to open one in program and the File Picker wouldn't show them.

AdrianEddy, 2022-08-07:
@shaheedmalik did you download latest dev build? Also braw is only supported in Motion data section, not the video section

shaheedmalik, 2022-08-07:
> @shaheedmalik did you download latest dev build? Also braw is only supported in Motion data section, not the video section

I downloaded the latest build and tried to open the Motion data section but it sat there with the spinning wheel.

AdrianEddy, 2022-08-07:
Yes, you need to load the video first