gyroflow/gyroflow #392: Video stabilization with unknown lens profile?

DavidGP, 2022-07-03:
Hello all, 

I hope it's ok if I ask my question here. I have not found a forum for Gyroflow yet.

I am trying to stabilize [this very shaky video](https://youtu.be/hgSJJlykSLI?t=39).

Since I am not the creator of this video, I don't have a lens profile for it. 

What would be the best settings to use in Gyroflow in this case?

Thanks!

AdrianEddy, 2022-07-03:
Do you know which camera and lens it was shot on? Gyroflow has a large database of lens profiles so there's a chance someone made a profile for this setup. Do you have gyro data? 

Also for questions like that use Discord https://discord.gg/WfxZZXjpke

DavidGP, 2022-07-03:
Thank you Adrian. I have posted the question at Discord!

closed 2022-07-03 by DavidGP