gyroflow/gyroflow #394: VAAPI on intel HD not working

ask6483, 2022-07-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.1.0 (de24082)

### What operating system are you using?

Debian GNU/Linux bookworm/sid

### What GPU are you using?

Intel HD 620 (KBL GT2)

### What happened?

VAAPI is not working, only CPU rendering available.

### Relevant log output

```shell
[swscaler @ 0x7fa127853100] 0bpp not supported by yuv2rgb
[swscaler @ 0x7fa127853100] vaapi is not supported as output pixel format
20:45:22 [DEBUG] (2) gyroflow::rendering: ffmpeg_hw::supported_gpu_backends: ["vdpau", "cuda", "vaapi", "qsv"]
20:45:22 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [dec] codec type AV_HWDEVICE_TYPE_VAAPI 1
20:45:22 [DEBUG] (2) gyroflow::rendering::ffmpeg_processor: Selected HW backend AV_HWDEVICE_TYPE_VAAPI (vaapi) with format Some(AV_PIX_FMT_VAAPI)
20:45:22 [DEBUG] (2) gyroflow::rendering: proc.gpu_device: Some("vaapi")
20:45:22 [DEBUG] (2) gyroflow::rendering: Possible encoders with Intel: [("h264_vaapi", true), ("h264_qsv", true), ("h264_v4l2m2m", true), ("libx264", false)]
20:45:22 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [enc] codec type AV_HWDEVICE_TYPE_VAAPI 0, for: h264_vaapi
[src/rendering/ffmpeg_hw.rs:174] &dev.hw_formats = [
    VAAPI,
]
[src/rendering/ffmpeg_hw.rs:175] &dev.sw_formats = [
    BGRA,
    ARGB,
    RGBA,
    ABGR,
    BGRZ,
    ZRGB,
    RGBZ,
    ZBGR,
    X2RGB10LE,
    GRAY8,
    NV12,
    YUYV422,
    UYVY422,
    YUV420P,
    YUV411P,
    YUV422P,
    YUV440P,
    YUV444P,
    P010LE,
]
20:45:22 [DEBUG] (2) gyroflow::rendering: video_codec: Some("h264_vaapi"), processing_order: PreConversion
20:45:22 [DEBUG] (2) gyroflow::rendering: start_us: 9133674, render_duration: 437616.3259215043, render_frame_count: 26257
20:45:22 [DEBUG] (2) gyroflow::rendering::ffmpeg_processor: Codec formats: [VAAPI]
20:45:24 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Hardware transfer formats from GPU: Some([NV12, BGRA, ARGB, RGBA, ABGR, BGRZ, ZRGB, RGBZ, ZBGR, X2RGB10LE, GRAY8, YUYV422, UYVY422, YUV420P, YUV420P, YUV411P, YUV422P, YUV440P, YUV444P, P010LE])
20:45:24 [DEBUG] (2) gyroflow::rendering: No matching codec, we need NV12 and supported are: [VAAPI]
```

AdrianEddy, 2022-11-18:
Does this still happen in latest dev build? https://gyroflow.xyz/devbuild/?autodownload

If yes, please post the log

AdrianEddy, 2022-11-20:
Duplicate of #246

closed 2022-11-20 by AdrianEddy