gyroflow/gyroflow #397: Non-symmetric lens option causes lower left zoom bias

avsaase, 2022-07-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

599ab50515087bce9ba39118a709973693a92047

### What operating system are you using?

Windows 10

### What happened?

With the new asymmetrical lens option, there is a zoom bias to the lower left:
[image]

When lens correction strength is set to 0%, black borders come into frame:
[image]

This example is from a video from the Runcam 5 Orange using the XV setting and corresponding lens profile but I also tried it with `"input_horizontal_stretch": 1.0,` and the dimensions as if it was a normal 4K video, with the same result.

Here's a clip and `.gyroflow` file with synced gyro data: https://drive.google.com/drive/folders/1II7j7cD8n304s1ZmVXIiKFHtL8jGUE0w?usp=sharing

avsaase, 2022-07-09:
Actually, I think the off-center zooming with 100% lens correction strength is not a bug and is just the result of the asymmetric lens. 

Stabilization toggled off:
[image]

Stabilization toggled on:
[image]

With zooming center offset X=10% and Y=-10% to use as much of the frame as possible and get the effect I expected when I opened this issue:
[image]

On the last still, there is a lot more FOV on the right and top but that's because the camera actually has more FOV in those directions (assuming the lens profile is correct). I guess the tolerances on this optics of the camera are terrible and everyone should make their own profile instead of using someone else's.

VladimirP1, 2022-07-22:
Can confirm with 100% correction strength.

[image]

Will upload the files soon

VladimirP1, 2022-07-22:
Here's the link to the files https://drive.google.com/drive/folders/1VSRTcXCLzYgNRkPW5h3ndOUrJ8LaPz2R?usp=sharing

VladimirP1, 2022-07-22:
I am stupid, I had no zooming selected

AdrianEddy, 2022-11-21:
Fixed in 024755d13fd9c3fadcc55006067b07f7782a04a4

closed 2022-11-21 by AdrianEddy