gyroflow/gyroflow #402: Rolling shutter correction works incorrectly for flipped videos (even with the checkbox checked)

VladimirP1, 2022-07-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

gh605

### What operating system are you using?

windows 11

### What happened?

Steps to reproduce:

- Stabilize a video in GyroFlow with horizon lock
- `ffmpeg -i video.mp4 -vf "transpose=2,transpose=2" -c:a copy -pix_fmt yuv420p video_flip.mp4`
- Load the video in GyroFlow, click 'guess gyro orientation', stabilize with horizon lock
- Compare the results

Expected: videos are the same
Actual: the flipped video has some wobble

Samples (both stabilized, unstabilized and project files): https://drive.google.com/drive/folders/1iUkXTg45m1cmxzb0deddWCuYLOrEGI-i?usp=sharing

AdrianEddy, 2022-07-14:
It works correctly, but the "bottom to top" checkbox still assumes left-to-right readout time. So if you simply rotate the video 180 degrees, it will be bottom-to-top and right-to-left, which is not supported. If you mirror the video vertically, then the checkbox works correctly as expected.

Now the question is how does Insta360 record and report it, because they have negative rolling shutter values in metadata (indicating bottom-to-top), but I'm not sure if it's rotated or mirrored. I don't have the camera to check

closed 2022-07-14 by AdrianEddy

VladimirP1, 2022-07-15:
Oh, I've never thought that line readout time (usually <5us) can be noticable

AdrianEddy, 2022-07-15:
> Oh, I've never thought that line readout time (usually <5us) can be noticable

Line readout time probably isn't, but if the readout is right-to-left (like in 180 deg rotated video), then we're correcting in the opposite direction, which is noticeable. Unless you're referring to something else?

VladimirP1, 2022-07-15:
Ok, now I understand