gyroflow/gyroflow #405: one session only (Linux)

nightflasher, 2022-07-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.1.0 (gh570)

### What operating system are you using?

ubuntu 22.04 LTS

### What GPU are you using?

AMD ATI Radeon R7 530 | INTEL UHD 620

### What happened?

After editing/stabilizing a video (runcam thumb v1), i can't open another video for editing. Have to close the Programm after every edited video.

JoshuaNitschke, 2022-07-26:
@AdrianEddy this is my next one, starting it next weekend.

JoshuaNitschke, 2022-07-31:
So i am able to reproduce the issue.  It seems you have to have started playing a video for the issue to occure as I can switch videos mulitple time.

Steps to reproduce
1) Load 
2) Click play and let the video play
3) Observe you cannot load another vidoe, it just hangs.

The process dies here:
https://github.com/AdrianEddy/qml-video-rs/blob/2c642501b7704ee278331f272f9c0efd66e630f4/src/video_player.rs#L41

AdrianEddy, 2022-07-31:
@wang-bin any idea about this?

nightflasher, 2022-08-02:
I need to play back the video, to see if the result satisfies my desire.

JoshuaNitschke, 2022-08-02:
@nightflasher obviously, I was just giving better details on how to reproduce the error so that @wang-bin can troubleshoot, the error is occurring from his library.

AdrianEddy, 2022-08-02:
actually there is one more function between mdk there:
```cpp
void MDKPlayer::setUrl(const QUrl &url) {
    m_overrideFps = 0.0;
    if (!m_item) {
        m_pendingUrl = url;
        return;
    }
    destroyPlayer();
    initPlayer();

    QString path = url.toLocalFile();
    m_player->setMedia(qUtf8Printable(path));
    m_player->prepare();
}
```

can you add logs here and check on which line it hangs there?

JoshuaNitschke, 2022-08-06:
@AdrianEddy - seems to have something to do with resetting the unique pointer?  I have no experience with C++ and I did some googling, but nothing really stood out to me.  I tried removing it and it seemed to let me switch videos a few times, but then I had the same issue again where it didn't progress passed the assignment `m_player = std::make_unique<mdk::Player>();`

[image]

AdrianEddy, 2022-08-07:
So the problem is in the destructor of `Player` in mdk-sdk. @wang-bin?

AdrianEddy, 2022-08-07:
@JoshuaNitschke can you try adding `m_player->waitFor(mdk::PlaybackState::Stopped);` in `stop()`?
[image]

JoshuaNitschke, 2022-08-07:
Yes, will do it later today 👍

JoshuaNitschke, 2022-08-08:
@AdrianEddy unfortunately the issue persists, although it seemed better?  
https://vimeo.com/737429743/7f4489e738

I added the wait on pause and play as well just for fun.  I have yet to see it fail to load a new video while the player is in play mode - might be worth digging into what is happening after clicking pause.

wang-bin, 2022-08-08:
gh570 crashes before showing the main window on my laptop, the log says no opencl. So I tested the nightly build, open a video, export, then open another file, it works fine

AdrianEddy, 2022-08-08:
It could be related to a particular distro or decoder. @JoshuaNitschke can you post entire gyroflow.log file? after this happens? we need to see which "video.decoder" mdk used etc. also what linux distro are you using?

AdrianEddy, 2022-08-09:
@nightflasher  and @JoshuaNitschke  please check the latest build https://gyroflow.xyz/devbuild/?autodownload

JoshuaNitschke, 2022-08-10:
@AdrianEddy I'll check on the weekend, sorry that I didn't have time to get you the logs - I checked your changes, you were able to verify it was an issue with releasing the pointer?

nightflasher, 2022-08-10:
AppImage doesn't work for me (dry start ./Gyroflow-linux64.AppImage and dedicated grfx card DRI_PRIME=1 ./Gyroflow-linux64.AppImage - see attached dumps) the extracted .tar.gz does work, but after playback of a video (loaded .gyroflow file) it crashes - no it locks up/freeze and i force quit.
[gyroflow_crash_dumps.zip](https://github.com/gyroflow/gyroflow/files/9303623/gyroflow_crash_dumps.zip)

AdrianEddy, 2022-11-18:
Does this still happen in latest dev build? https://gyroflow.xyz/devbuild/?autodownload

closed 2022-11-20 by AdrianEddy