gyroflow/gyroflow #406: Errormessage with export

HarDuenn, 2022-07-21:
Hi everyone,
I am trying the gyroflow software and get an error message when I start the export. That's in version 1.1 and the older ones.

What am I doing wrong or what else I do have to install?

Thanks very much

Harald

[Error.pdf](https://github.com/gyroflow/gyroflow/files/9161266/Error.pdf)

AdrianEddy, 2022-07-21:
Your GPU doesn't support hardware encoding. If you have two GPUs (common in laptops), you can try selecting the other one under Export -> Advanced -> Device for rendering.

Otherwise, you just have to render on the CPU, just uncheck "Use GPU encoding"

closed 2022-07-21 by AdrianEddy

HarDuenn, 2022-07-21:
Thanks very much.Now it works very fine!Best wishesHarald--Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.