gyroflow/gyroflow #407: Motion Data overlay

350d, 2022-07-22:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Hello! Can't find it in docs, but I can't enable to view motion data from external txt file over existing motion data built in video file. Only if I start auto sync - I can see fragments of this new data. The problem is that my logger start to log before video was started to record and I don't have exact time of this delay. I guess it will be very useful to see both motion data sources on one timeline to move it one over other and visually find approximate gyro offset and IMU orientation.

Im using Hero 7 files with blackbox gyro data recorded on external logger (sd card on uart)

AdrianEddy, 2022-07-22:
I'm not sure I understand. The optical flow is shown on the chart only on parts that were analyzed. Gyroflow doesn't analyze the entire video.
For your use case, you can just increase Sync search size, and it should just work.
You can also always add a manual sync point and set the offset manually to something large (like 300000ms) and then use the slider to move the data around to match it visually on the chart, with optical flow data.

350d, 2022-07-22:
I've tried SSS = 120s, max sync points = 3, every 1 frame, analyze per sync point - 6s
and its looks not even close to any true points. 

In my use case i need to overlay original GoPro gyro data with my blackbox data to visually sync it roughly.

AdrianEddy, 2022-07-22:
Overlay is not supported and it's not planned. I'm pretty sure it works, you can send me the video files and I can check it for you

350d, 2022-07-22:
Im sorry! Wrong video file was used... My fault! 
Anyway, you still can use existing GoPro motion data for auto sync in some way (additional source of motion data). Its. just to noisy, but still useful I guess.

AdrianEddy, 2022-07-22:
Could be useful, but it's not a trivial feature to implement and its use case is pretty niche. Optical flow based sync works very well and is enough in my opinion.

closed 2022-07-22 by AdrianEddy

350d, 2022-07-24:
@AdrianEddy Can you please help me anyway with my files. Here is the [short clip and blackbox log files](https://mega.nz/folder/6M5jlQ5Q#SpeRSxZCVC_F7oiseSr2nQ), auto sync works just fine but video still shaking for some reason. It looks more shaky with stab than without.
Maybe im missing something on my setup or settings? Appreciate your help here. Thanks!

AdrianEddy, 2022-07-24:
Did you set the Pitch rotation? looks like it's 17 degrees.
It works mostly fine, the remaining shakes are a hardware rather than a problem in Gyroflow - either frame vibrations, rigidity of FC vs the camera, noise from motors or stuff like that. I'm not good with hardware so you may have a better luck asking about this on Discord.

Here's the project file that works for me:
[GX010786.gyroflow.zip](https://github.com/gyroflow/gyroflow/files/9176699/GX010786.gyroflow.zip)