gyroflow/gyroflow #409: Cannot parse some of the blackbox data from betaflight

theysp, 2022-07-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.1.0

### What operating system are you using?

windows 10

### What GPU are you using?

RTX2070

### What happened?

Some of my blackbox log file cannot be parsed properly by gyroflow as motion data, only plat lines appear in the motion plot widget beneath the video display widget.
I have tried open this log file in Betaflight Blackbox Explorer, everything seems right.
May you have time look into it. 

[LOG005.TXT](https://github.com/gyroflow/gyroflow/files/9177417/LOG005.TXT)

theysp, 2022-07-25:
The log files are generated on betaflight 4.3, the same as following one which can be parsed well.
[LOG006.TXT](https://github.com/gyroflow/gyroflow/files/9177422/LOG006.TXT)

AdrianEddy, 2022-07-25:
Thanks for test files, as a workaround for now, you can export CSV from the blackbox explorer and load that csv to Gyroflow

AdrianEddy, 2022-07-27:
It's fixed in latest dev build: https://gyroflow.xyz/devbuild/?autodownload

These files are damaged however, there is a garbage in LOG005.TXT 1:52 to 1:53 and in LOG006.TXT at 1:13

I'm not sure what caused this damage, but it might be faulty flash chip or sd card or some other hardware problem. 
Nonetheless Gyroflow can now ignore this damaged part and continue parsing the rest

closed 2022-07-27 by AdrianEddy

theysp, 2022-07-28:
great, thanks